bambulab/BambuStudio #876: Parameter Table can no longer switch filament

kartoffelfighter, 2022-12-17:
**Describe the bug**
With previous versions, you could switch the used filament via the parameter table. In the latest version, the feature is buggy. Filament switch does no longer get applied

**3mf File for This Bug**
_independent of file_

**To Reproduce**
Steps to reproduce the behavior:
1. Open .stl
2. Split in half
3. Open parameter table via right-click 
4. Change filament for a part via drop-down
5. Click gets applied, but filament not

**Expected behavior**
Like before, the Slicer should change to the different filament

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.4.0.17

Sirwill1968, 2022-12-17:
Confirmed this fails on Windows, but works on Mac OS.

SaltWei, 2023-01-11:
same with https://github.com/bambulab/BambuStudio/issues/961 and will be fixed in next release

kartoffelfighter, 2023-01-20:
Updated the Bambu Studio today, works again in Version 1.4.2.13
Thanks!

closed 2023-01-20 by kartoffelfighter