bambulab/BambuStudio #878: Will there be custom printer/non network printer options? (Cr10,cr10v2-v3 or any custom printer)

poolboy97, 2022-12-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Benderrodrigu3z, 2022-12-20:
your asking for them to make a Creality machine.... that's a different companies crappy machine, why on earth would they ever want to downgrade themselves like that?

Admin close this for no comments on there own comment field

poolboy97, 2022-12-20:
Well they have ender as list of printers and like many others (ultimaker, prusa slicer etc.) you can setup non network printers. So i wouldn't thought/think this is completely ignorant question.

SaltWei, 2023-03-02:
It's open and welcome to add other printer preset. Since this is open source slicer.
But we don't have time to maintain them, and also we don't have all of these printers.
Ender 3 and voron have been now added already.

closed 2023-03-02 by SaltWei