bambulab/BambuStudio #880: V1.4 - Seam and Support Painting Causes Plate to Vanish - Mac

bonjipoo, 2022-12-18:
When you toggle seam or support painting tools in Bambu Studio v1.4 the plate disappears on Mac version.

[image]

macOS Big Sur 
MacBook Pro M1

lanewei120, 2022-12-20:
@bonjipoo 

the plate is not displayed under all of the painting pages
this is the expected behavior

bonjipoo, 2022-12-20:
@lanewei120 Sorry I do not understand.  Before version 1.4 I could paint a seem now I don't understand where the plate goes.  Please explain.

bonjipoo, 2022-12-20:
https://user-images.githubusercontent.com/5507101/208756115-cd9c50bb-9b78-4561-a3d5-d3682473c7be.mov

lanewei120, 2022-12-21:
the plate I said before means the bed-plate
so I missed your meaning

from your video, the object disappears

could you share your 3mf to us for a check

thanks

bonjipoo, 2022-12-21:
[3dbenchy.3mf.zip](https://github.com/bambulab/BambuStudio/files/10279998/3dbenchy.3mf.zip)

I had to Zip the file.  The model shown in the video above is not publicly available so I have uploaded the 3mf with a benchy.  Same issue with this file.

Haidiye01, 2024-01-15:
May I ask if this issue still exists with the latest version V1.8 of the software

Haidiye00, 2024-05-09:
After testing version 1.9, we did not find this issue. If you still have issues, please raise the bug again. Thank you~

closed 2024-05-09 by Haidiye00