bambulab/BambuStudio #881: V1.4 - Bambu Studio throws "error loading shaders" when opened - Mac

bonjipoo, 2022-12-18:
As soon as the application is opened. The error is displayed

[image]

macOS Big Sur
MacBook Pro M1

lanewei120, 2022-12-20:
@bonjipoo 
hello, what is the detailed version number of your machine?
please help to send the log files in /Users/user.name/Library/Application Support/BambuStudio/log here

thanks

lgbell, 2022-12-26:
Have the same issue- error loading shaders-- BambuLab Studio 1.4.0.18, MacOS BigSur 11.6 MacBook Pro M1 chip- couldn't find the log file according to the listed path to send to support

bonjipoo, 2022-12-26:
I wasn't able to find the log file requested.

I also wonder if the shader error is the problem bedhind the seam and support painting issue on MacOS BigSur with M1 chip

lanewei120, 2022-12-27:
hello

@bonjipoo @lgbell 
we have fixed this issue and released the new version v1.4.1
please have a try

thanks

closed 2023-03-02 by SaltWei