bambulab/BambuStudio #882: Auto-calc multiplier not working

mmontminy, 2022-12-18:
**Describe the bug**
The auto-calc multiplier has changed. I used to regularly use .45 as a multiplier. Now that appears to do nothing. It looks like it might be changing the values but not updating the display. If you use a multiplier that drives the numbers out of range you get an error, but the shown values never change.

**3mf File for This Bug**
No special file needed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Flushing volumes'
2. Click on 'Multiplier', enter a new value other than 1.0
3. Click "Auto-calc'
4. Values don't change/update

**Expected behavior**
When you enter a new multiplier, auto-calc should update the flushing volumes.

**Screenshots**
The default multiplier:
[image]
Changing to 0.3 results in:
[image]
Changing to 0.4 results in:
[image]
It's doing something, but the values aren't changing...
Increasing the multiplier also doesn't appear to increase the numbers:
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10
 - Studio 1.4.0.17

zhimin-zeng-bambulab, 2022-12-29:
In general, use 1.0 as multiplier can meet the requirements.  The multiplier is used for additional scaling when the user thinks that the automatically calculated value is too small or too large.

mmontminy, 2022-12-29:
1.0 is too aggressive. It also doesn’t address the fact that the scaling factor used to work and now doesn’t.

zhimin-zeng-bambulab, 2023-01-18:
We will optimize the interaction behavior in future version.

closed 2023-03-02 by SaltWei