bambulab/BambuStudio #883: New Send feature not working, wrong access code error

mmontminy, 2022-12-18:
**Describe the bug**
Trying to use the new Send and Send all fail with an error "wrong access code. Please log out and log in to the printer again".
I've tried re-logging to the printer with no change.

**3mf File for This Bug**
Any file will cause the error.

**To Reproduce**
Steps to reproduce the behavior:
1. Select "Send" or "Send all" after slicing
2. Click on 'Send....'
3. After a brief progress bar an error is displayed

**Expected behavior**
I expected the file to be uploaded to the printer's storage so it could be printed from the control panel of the printer.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10
 - Studio 1.4.0.17

GuyH77, 2022-12-18:
I had this, just do as it says and log out and back in on the printer. It sorted it, working fine since.

mmontminy, 2022-12-18:
> I had this, just do as it says and log out and back in on the printer. It sorted it, working fine since.

I have, several times now. I just tried again and it's the same.

FreeManFreeThinker, 2022-12-19:
I have this issue as well. macOS 13

chrishilbert, 2022-12-30:
Also have this issue.

Merlwynd, 2022-12-30:
There is more than one issue with sending a file to the printer SD card. See comment 982. There have also been comments in various forums. So far Bambu has remained silent.....I wish they would see it as a higher priority...or it least give some guidelines on what to try to fix......

walterwongbbl, 2023-01-16:
We will try to reproduce this problem. are your printer and studio on the same network?

walterwongbbl, 2023-01-16:
HI, When wrong access code is prompted again, a LAN connection dialog will pop up, allowing you to manually enter the printer's IP address and access code to connect. （next release version）

SaltWei, 2023-03-02:
V1.5.0 public beta has been released

closed 2023-03-02 by SaltWei