bambulab/BambuStudio #884: Not slicing when "Engineering plate" and PLA filament selected

ClauMat, 2022-12-18:
Bambu Studio does not slice when you select "Engineering plate" and you want to use PLA or any custom filament profile.

Many users have adhesion problems without receiving a concrete solution from Bambulab to this problem.
For me at least, the PLA I use only sticks to the "Engineering plate" at 65 celsius.

It's up to me on what surface I print and with what filament, the slicer should not limit this.

[image]

 - OS: Windows 10
 - Version: 1.4.0.17

ashleyuk90, 2022-12-18:
This release change a bunch of allowed filament / plate types inline with what they have found, due to some not sticking well or some sticking too well and removing the top surface.

If you check the store on the plates your now see what each plate is supported, this is now inline with the allowed list within Studio/FW.

If you want to bypass the suggested / support combinations your just need to lie to Studio what plate your using.

ClauMat, 2022-12-18:
> If you want to bypass the suggested / support combinations your just need to lie to Studio what plate your using.

The problem is that it tries to limit something. Something that you can bypass in several ways anyway.
I ask you, what is the purpose of such a limitation?

SaltWei, 2022-12-29:
This is not a bug tickets.
Also this limitation is not designed by Studio but by the designer of plate.
The limitation can help newbies a lot to avoid sticking issues.

We will evaluate one new option so user can choose not showing these warnings anymore.

SaltWei, 2022-12-29:
By the way, you can edit temeperature by youself to make it to be non-zero value. And the limitation will disappear when slicing.
[image]

SaltWei, 2023-02-09:
For next version, the pop-up message will be changed as following.
[image]
User can unfreeze this limitation easily, but we still keep this limitation when using default settings to avoid sticking issues for newbies.

StuTheDog, 2023-06-25:
My model is using PETG for both model and support (on build plate) with PLA as the interface.  However am getting a message stating the build plate exceeds the filament's temp and needs the front door opened.  Please ask the developers to not display this dialog when none of the filaments that contact the build plate have this issue.

SaltWei, 2023-06-25:
> My model is using PETG for both model and support (on Engineering plate) with PLA as the interface. However am getting a message stating the build plate exceeds the filament's temp and needs the front door opened. Please ask the developers to not display this dialog when none of the filaments that contact the build plate have this issue.

Good suggestion. Only check the bed temperature when the filament is contacted with the bed.
We will try this improvement