bambulab/BambuStudio #886: Print head collides with printed object when using 'print sequence: by object'

kvnper, 2022-12-18:
BS Version: 1.4.0.17
OS: Windows 10

Brief: Print head collides with printed object when using 'print sequence: by object'.

Description: The print head collides with the first printed object when printing one object at a time.

Project file attached:
[per-object-issue_BS1.4.3mf.zip](https://github.com/bambulab/BambuStudio/files/10253942/per-object-issue_BS1.4.3mf.zip)

Here is a screenshot of the positioning of the objects, showing the collision(?) avoidance area. The object on the right will print first and will be collided with when the second object begins printing.
[image]

nkacena, 2023-01-23:
also having this issue. it seems print sequence is inherently flawed in that when it goes to change filaments and then resume a print, the print head does not account for previous object height so its path back to resume printing follows the shortest path and collides with a previous print.

SaltWei, 2023-03-02:
Please use v1.5.0 Public beta to have a try.
Have fixed the collision issue of changing filemant when printed by object

closed 2023-03-02 by SaltWei