bambulab/BambuStudio #888: version 1.4 no send button no klipper selection for ender 3 v2

Qballjos, 2022-12-18:
**Describe the bug**
After updating to version 1.4.0.17 no more klipper firmware selection for ender 3 v2, app crash when trying to add physical printer (ipadres), send button disappears 
**3mf File for This Bug**
Not related to slicing

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

https://user-images.githubusercontent.com/1911646/208310074-37525241-59bc-4323-8411-b5be7be83cee.mov

**Desktop (please complete the following information):**
 - OS: Macos 
 - Version 13.1

lanewei120, 2022-12-20:
@Qballjos 

hello
the send function need the firmware support
so for third-party printers
only "Export G-Code file" can be used

thanks

Qballjos, 2022-12-20:
Well this did work with previous version until they decide to remove the klipper firmware portion for ender 3 v2 again.

lanewei120, 2022-12-21:
oops, correct my previous comment
the send function is also available on third-party printers which supports remote print

we will check it

also does your ender3's firmware the same with before?

Qballjos, 2022-12-21:
I have klipper running on my Ender (latest version) This option was available to select in the previous version.

StoneLiBambulab, 2023-01-11:
it is removed by mistake. i will add it back.

Stone Li

StoneLiBambulab, 2023-01-15:
@Qballjos ,
 Hi, I have fixed the bug and the version will be published next week. The "Send" will be moved in the the print dialog . And the wiki page will be published after reviewed, https://wiki.bambulab.com/en/software/bambu-studio/printing-gcode-though-octoprint 

Thanks!
Stone Li

Qballjos, 2023-01-21:
Ok the print option works again that's perfect, but the option to select klipper as firmware is still gone...., and the device tap no longer opens the mainsail/fluid page.

StoneLiBambulab, 2023-01-29:
Hi @Qballjos ,
These commands are not supported by klipper, and it should be skipped and will not affect printing.

Thanks!
Stone Li

closed 2023-03-02 by SaltWei