bambulab/BambuStudio #889: [The "send to printer" function does not work as expected.] "Send to Printer" not working

BambuLabDiscord, 2022-12-18:
When trying to send a plate or multiple plates to my X1-C, I always get the error message "Send to Printer failed. Please try again".

miloutintin, 2022-12-18:
I, too, get this almost every time.  Usually one retry is needed, but sometimes several retries is required before it will load to the cloud.

UndeadKernel, 2022-12-19:
I reported this issue through Discord.
In my case, no matter how many times I try to "Send", I always get an error message.

UndeadKernel, 2022-12-29:
I'm still observing this issue.
I have now tried using a different windows machine in the same LAN, and I have deactivated the windows firewall. Still, I cannot send anything to the printer.

Is there an easy way to debug what the problem could be?

james7360, 2022-12-29:
I found a solution on the bambulab forums by obicom66. 

"Unfortunately I can’t see the message but on the beginning I got an error in regards of wrong access key. Logoff & Login again to the PRINTER (not Bambu Studio) solved the issue for me. Now I can use the send option without any problem."

This worked for me just re logged in to the printer using the qr code in the app and then restarted Bambu Studio.

UndeadKernel, 2022-12-29:
That did the trick @james7360. After logging off and on to the printer, I can finally send files to it.

Thanks for the tip!

closed 2023-03-06 by walterwongbbl