bambulab/BambuStudio #890: STOP DISALLOWING PRINTING

AbeFM, 2022-12-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Bambu will decide something isn't possible. And it will just refuse to print.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Pop up a WARNING, with an option to ignore/print anyway

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

I've been printing TPU without a SINGLE issue since I got the machine months ago. Now, I have to make ANOTHER material in my list of fake materials. My user experience is "Do something that already works. Now, go in circles trying to TRICK it into working. Deal with the consequences of making it work."

Now I'm hand-tuning stuff that already worked to try to make it go. A warning, like the vitrification temp, makes sense. But now I can't use the printer to print stuff where it would print the exact part before. It really is upsetting that I need to do so much work. This specific case is "TPU", which the AMS let me load, but will not let me use. There is NO logic in that.

It's happening over and over. It happens with the bed surface, so I have to have special fake materials to print PETg on the engineering plate, and it makes the process of loading filament more of a pain because I have to tell the machine it's PLA, then make fake PLA profiles that are actually petg.

Intelligence is great to build into the product, but it is rendering the printer unusable. It's not the machine being incapable, it's just like you came here personally and turned the machine off right after I hit print. Please allow advanced users to BYPASS warnings.

Benderrodrigu3z, 2022-12-20:
Also you call yourself advance but you are clearly NOT advanced, learn the machine, don't do a it worked once it should work again type of mentality.

Benderrodrigu3z, 2022-12-20:
what would they call this feature? turn on id10t mode?

AbeFM, 2022-12-21:
Sigh. Not adding much there.

I started with "I've been using TPU regularly, without issue, for months".

There is no point in telling me you're scared to try something. I've got shelves full of things printed with TPU, fed through the AMS, and all I want is to return the capability to do so.

Other things were disallowed by the machine and are now allowed.

By way of example, Bambu tech support specifically told me to use the other sheet, and told me to ignore the warnings generated by the slicer.

Please allow Bambu to respond, you're not qualified to comment on what they are doing and why. I'm happy to have a philosophical discussion on the forums of discord, but do not muddy the waters with your Fear, Uncertainty and Doubt.

This is a place to submit feature requests. I understand that you guys want to tell me how I should see the world, but your advice about my mental state is not relevant.

Bambu has been responsive in the past, please allow them to continue to consider suggestions in light of their INSIDER knowledge, not an outsider's assumptions.

Benderrodrigu3z, 2022-12-21:
Whats it going to be called? Override for Dummies?

Benderrodrigu3z, 2022-12-21:
The Warranty Voider 3000 ?

AbeFM, 2022-12-21:
You guys are super helpful.

Please get off the platform. You could well have commented on my tone, but
instead you just opened your mouth and started letting noises come out. I'm
not going to explain it to you again.

rmhenn, 2022-12-21:
Just because you've been using TPU for months doesn't mean anything to the fact that the AMS is not intended to use that material. It's legitimately on the first page of the quickstart guide that is taped to your printer when you unbox it (in bright red letters too). Whether or not I'm "scared to try something" is irrelevant to the fact that this is not recommended by Bambu to the extent that they are locking out it's use with the AMS in the slicer, and asking for them to allow you to bypass the lockout is honestly akin to asking car manufacturers to allow you to bypass the airbags.

As for previously disallowed features being allowed now, yeah, sure. This most recent update integrated some user feedback and optimizations into build plate selection and filament compatibility so it's not surprising that some more options became available, but that doesn't preclude the fact that you shouldn't be using TPU in the AMS and all that they've done is ensured you can't break your device through improper use.

As for going after my qualifications and accusing me of FUD, all I'm telling you is what the manual says (the first page of it at that) and if you somehow think that's not information from the manufacturer then I can't really help you.

Benderrodrigu3z, 2022-12-21:
> You guys are super helpful. Fingers in your ears. Please get off my thread. You could well have commented on my tone, but instead you just opened your mouth and started letting noises come out. I'm not going to explain it to you again. You're not ADDING to the discussion, but you're letting everyone know just how valuable your input can be. Grow up.
> […](#)

since when did typing become speaking?
Let the troll war begin, get in here Discord

Photogad, 2022-12-21:
I see no problem with giving the user the ability to dismiss warnings, as long as they were warned. +1. In America we ❤️ FREeEDOM 🇺🇲

AbeFM, 2022-12-21:
These guys are turning this into a social networking farce.

AbeFM, 2022-12-21:
Ah, muted the clowns.

Bambu:
This is not the only thread discussing the topic of non-bypassable "warnings". It is a general improvement which is needed of which this is a poignant example.

The request fits with, for instance, the recommendation to create fake materials which came straight from Bambu.

It now goes without saying that some people are confused thinking the capabilities of the machine extend no further than the default settings.

harriettubgirl, 2022-12-22:
The way you're going about this is so wrong... Just open a new thread with a feature request to unlock material locks on the ams. They don't want to deal with potential breaks from using TPU in the AMS, they obviously found it's not reliable. Just because you've made it work before doesn't mean that much sorry.

databonanza, 2022-12-24:
Another end users that thinks they're smarter than a team of engineers that are on payroll.

Just because you got your car to operate without air bags doesn't mean ....... blabhblhablablhab.

AbeFM, 2022-12-24:
Ah, spew unhelpful nonsense then delete your posts so people won't see them.

[image]

You want a flame war. Except if people will see you asking for it, apparently.

All I want is for the printer to work well. You aren't helping the devs to do their job by starting a bunch of fights in their back yard. If you have something meaningful to add about the product, go ahead.

Harriet was *helpful* with her comment.
Photogad actually agrees with my request
Bambu tech support agrees that there are many things which aren't yet perfected in the system, and they TELL me to work around the shortcomings, to ignore warnings generated by the device, etc.

I'm going to start another thread on the topic, and I hope you have the maturity not to piss in that pool too.

closed 2022-12-24 by AbeFM