bambulab/BambuStudio #892: Suggestion: Cut and continue print before full purge

JPotter73, 2022-12-18:
Not a current problem but a suggestion

When printing multicolour, a good portion of the waste is from the left over cut off piece still in the hotend that needs to be pushed out for the new colour. That portion is still nice and clean. 

Instead of dumping that out as purge, could it simply be estimated how much is there to finish off the last bit of the layer in that color and use it up in the print?

For instance, a 2 colour print of White and Black,

Currently:
The layer starts with white.
Change to black happens after all the white is fully printed out:
->Cut white and pull back to AMS
->Black loads in behind the white and purges the cut off white and what's needed to transition to clean black.
Black finishes the layer and starts the next layer

My alternative
The layer starts with white
Change to black happens before the white finishes printing out:
->Cut white and pull back to AMS with 75% of the cut distance left to print in white.
->Black loads in behind the white and prints the rest of the white off for the layer, leaving only 25% of the white cutoff to purge.
->After the white finishes, flush 25% of the white and what is required to transition to clean black. 
Black finishes the layer and starts the next.

Benderrodrigu3z, 2022-12-20:
the issue there is that to extrude you need to be able to retract, you cant retract it past the cutter so you would get globs, all it could to is push the rest of the filament through with more filament which would get in the hotted and need to be purged out.. your in a loop

JPotter73, 2022-12-22:
Agreed. Didn't consider retraction. Closing this request

Thanks!

closed 2022-12-22 by JPotter73