bambulab/BambuStudio #893: [BUG] Slicer set the bed type to null and crash

H4kunaM4tata, 2022-12-18:
My slicer started crashing out of the blue for no reason. I am still using the old release.
So I made all the changes, exported the project and imported in into the latest slicer release.

Even so the latest Christmas release doesn't display the plate due to tons of problems with Linux version ( Linux users might finally get attention by the year 3000 for Christ sake ) it was slicing fine although I couldn't see s**t

By doing it so it complained about Cool Plate which I do not use, but by checking the slicer that was crashing I noticed that the bed type was set to null. There was no bed type.

I cannot replicated it, after selecting PEI sheet, it started working again and I cannot not select a bed type so this was clearly some weird bug.

For what it is worth it, the slicer should not crash but rather send a dialogue about it.
I only use PEI sheet and never ever change it so there was no reason for me to even look at the bed type to noticed it empty. that took me ages to identify the problem.

Thank you

[image]

lanewei120, 2022-12-19:
do you mean the crash happened on your linux slicer?
what is the detailed steps?

it seems your report serveral issues
could you describe them one by one following the bug report format
[image]

H4kunaM4tata, 2022-12-19:
Like I mentioned before, there is no way to replicate it, I cannot set the bed type to null myself.
The slicer tho did set the bed to null causing it to crash.

That exposes the fact that the slicer would rather crash than warn you no sheet was selected.

lanewei120, 2022-12-19:
so how do you know it was crashed on a null bedtype?
any callstack on linux?

anymore, it seems you are building a linux version by yourself
is it possible for you to try the build following this [linux build guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)?

we have verified on ubuntu and fedora, both work fine without your issues.

H4kunaM4tata, 2022-12-19:
Because the bed type was blank and as I said twice, I cannot just set it to blank.
After setting bed type from blank to PEI it no longer crashed.
The slicer did set it to blank but I cannot. 

@lanewei120, Linux is not composed of just Fedora and Ubuntu.
An AppImage has to work with different system just like Cura and PrusaSlicer always worked.
I never saw such problems with those slicers.

I believe you spent at least 5 minutes reading here https://github.com/bambulab/BambuStudio/issues/12 to realize that the slicer for Linux isn't just completely broken but there is no help whatsoever from BambuLab team.
I mean, there is no even an official Linux slicer release like Cura and PrusaSlicer.
Users are helping each other to have it somewhat working. 
Where is BL team?????

For what matters it is not working on Ubuntu either because I checked.

The latest slicer version is broken with in so many different Linux OS
If that isn't bad enough, the new firmware seems to be incompatible with the old slicer version as you can see here https://github.com/bambulab/BambuStudio/issues/898

So before replying with "things are working" please, spend just 5 minutes reading that post to see that things are far from okay.
Thank you

lanewei120, 2022-12-19:
currently we don't have the linux formal release and we are doing it in the comming months
just as I explained in that issue [#12]( https://github.com/bambulab/BambuStudio/issues/12#issuecomment-1336581365)

as you said linux has so much versions,
and our engineers have not worked on linux before, it is not an easy job for us to finish

for fedora and ubuntu
what I mean is that we tried the studio built from [linux build guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide), it works well here
however, we just tested it on the only computer here which installed fedora/ubuntu
I just suggest you to try the building following that guide for ubuntu too

now as we can not reproduce you rissue, and we don't have the same environment,
please help to upload log for those issues
and I think we need to resolve them one by one, and for each issue
give more information and log

thanks

H4kunaM4tata, 2022-12-22:
@lanewei120 Linux slicer is now in a working condition for must of us as you can see on that Linux tread.
A small change is still needed to be merged to master and was the missing bit preventing the plate to show.

As for the missing bed type, I guess I will need to keep a closer look into it.

For now it makes sense to close this ticket, but it makes even more sense to change the slicer behavior, if there is a problem, notify instead of just close.

closed 2022-12-22 by H4kunaM4tata