bambulab/BambuStudio #895: Ability to apply adaptive layer height to multiple objects at once

FreeManFreeThinker, 2022-12-19:
**Is your feature request related to a problem? Please describe.**
It may identify an oversight that has been overlooked.

**Describe the solution you'd like**
The option to apply adaptive layer height to multiple objects or entire plates simultaneously.

**Describe alternatives you've considered**
I have to manually apply the layer height option to each object individually.

ashleyuk90, 2022-12-19:
I think this may be difficult, as adaptative layer height looks at the geometry of each object and creates the custom layer height mapping based on this, therefore the display of the layer heights and ability to adjust wouldn't work as each object would be different.

Benderrodrigu3z, 2022-12-20:
> What about the case when you are printing 50 times the same object on the plate, I could really see this useful when using clones, somekind of hierarchy inheritance maybe

I mean why not just apply the adaptive to one of them and copy it 49 times....

jukefr, 2022-12-27:
I was going to open a new issue but figured I might as well put it here

**Is your feature request related to a problem? Please describe.**
Not really, currently I have to select every object one by one and enable adaptive layer height.

**Describe the solution you'd like**
A global option (maybe in the `Other` section) to enable automatic adaptive layer height on every object.

**Describe alternatives you've considered**
Simply doing it manually for now

magic-happenz, 2023-05-02:
I came to this issue while I was looking for adaptive layer height when working with multiple plates as this does not seem to work either. When you select an object on another plate you simply see not any results so I assume it won't really work.

ChrisRArendt, 2023-11-13:
Provide this feature please

JesperKrogPoulsen, 2024-07-13:
Yes, please. I was looking for this. I have 1 object where I apply variable layer height and select Adaptive. It is an object with multiple colors and I have other objects that are internal to the model, so I want to use these as flush objects. If I add them, I can't use the prime tower, becuase it is only supported if all objects have the same variable layer height. I can't select all objects, because I can only apply variable layer height to one objects at a time. I can't merge the objects into one object, because then I can't use the other objects as flush objects. So a "clone layer heights" or the ability to apply layer heights to multiple objects at a time is welcome.

pzirnik, 2024-08-06:
I came to exactly the same issue. I like to print a model of a car. It has slightly differences between left and right side of the car, like gas tank lid etc. Also for printing i split the car model to left and right part in the middle to glue it together later. In the fender area i want to apply adaptive layer height, which should be identical between left and right part. When i do not get it identical, i am not able to enable prime tower as this does require adaptive layers between all parts on the plate. I even wonder how slicing is possible if objects do have different layer height on one plate ? For one plate all objects "must" share the same layer height, or not ? Only exception would be when i change print sequence to "by object".

pzirnik, 2024-08-08:
Forget it, found the solution.
Solution is to merge all objects on the plate, then you can apply variable layer height to all objects at once.

SREallyMatt, 2024-11-06:
> Forget it, found the solution. Solution is to merge all objects on the plate, then you can apply variable layer height to all objects at once.

While this works for single color prints, this doesn't work for multi color prints with a purge object. In addition this also prevents you from using the "skip object" functionality in case one of many fails during printing

HowardTomlinson, 2024-11-13:
+1 here for this capability, here's my use case.
I'm printing multiple objects where I use adaptive layers to be thicker in places where there are colour changes, in order to reduce the number of changes. These models are mostly white, and purging to infill on the models shows through.
 In addition I'm adding some _**different**_ objects to the plate which which I'd like to use to put purges into - both infill and model. These are items that will get painted.

So I can group up all the objects, apply the adaptive layer to the group, but then I can't select just the purge objects to set them as purge. If you 'split' a merged object where you've applied adaptive layers, all of the objects get their layers reset.  Conversely, I can't apply the adaptive layers to the purge object separately as there's no way to copy+paste them from one to another.

I *can* turn off the prime tower, and just trust the purge objects to do the same job. The order of setting properties is important here - you can assign purge objects only when the prime tower is enabled. When you disable the prime tower the purge objects still keep their property. Studio won't let you do it the other way round.

TheoRK667, 2025-01-06:
+1 from me. I have a Wallet to print, outer shell and sliding part print in place, cannot use adaptive layers because both need to be the same and I cannot get it calculated globally or even copy from the object that needs it to the object that does not care (shell) to harmonize.

TechlyAccurate, 2025-02-10:
please support applying adaptive layer height from one object to all other objects on the plate

this is, in my case, similar to several others; for the purpose of 'purge into object' which can only be done if the purge tower is enabled and the purge tower can only be enabled if the layer heights are the same for all objects.

Seeing as how I don't care if my purge doesn't have ideal layer heights, Im fine with giving it the same layers as the primary print

Unfortunately currently there is no way to apply the adaptive layer height globally, or copy it from one object onto another and copying the object doesn't work because when you replace the STL it loses its adaptive layer heights.