bambulab/BambuStudio #896: v1.4.0.17 No longer retains previously used Printer/Process preset

Ming-MCS, 2022-12-19:
**Describe the bug**
The new update of the slicer (1.4.0.17) no longer remembers the previously used presets in the Printer and Process categories which was present v1.3.1.1. Reopening the slicer now will switch the Printer preset to the system preset, instead of the one I had previously selected

**To Reproduce**
Steps to reproduce the behavior:
1. Open the slicer
2. Switch Printer/Process preset
3. Do a print (or not)
4. Exit application
5. Reopen application

**Expected behavior**
In v1.3.1.1, whenever the presets were switched (I have custom presets made for both Printer, Filament, and Process categories) and the application was closed, reopening the application would retain the Printer and Process presets that were selected before closing (Filament does not do this, and it would be great if it did as well). This would mean that every time I exported from Fusion360 directly into the slicer, I would only need to change the filament type and send it off to print. In v1.4.0.17, The Printer preset is now automatically defaulted to the system preset, which in turn also resets the Process preset to the default system preset, meaning I would now need to switch the Printer, Filament, and Process presets to the desired one before printing

 - OS: Windows 10 22H2
 - Slicer: v1.4.0.17

bambu123, 2022-12-19:
Please help to do these steps:
1. Rename %APPDATA%\BambuStudio to BambuStudio1
2. Open Bambu Studio (need re-login)
3. Do something to see if this issue reproduce
4. If it doesn't reproduce, please zip BambuStudio1 and send it to us (you can remove directory plugins to reduce package size)

Ming-MCS, 2022-12-19:
It's working now after renaming the folder and letting the slicer regenerate the configurations. Must've been some bloat in the configs from older versions of the slicer, but it's all good now.

Thanks

closed 2022-12-19 by Ming-MCS