bambulab/BambuStudio #901: [Feature request] Being Able to set a min and max layer height in the adaptative layering menu

NairoDorian, 2022-12-19:
I was trying to use the latest implementation of the Adaptative Layer Menu, but unfortunately while trying to setup the layering for my parts I reached a layer height too small and the melted filament is somehow too close to the previous layer and causes problems.

A quick and very useful fix would be to be able to setup a  minimum and a maximum layer height value for the adaptative layer function.

FlowForm, 2022-12-19:
Yes, this function is very needed. Very much like Cura does it.

cameo69, 2022-12-19:
I second that.

cann0nf0dder69, 2022-12-23:
No need to reinvent the wheel, it works well in cura, adopt what they're doing instead of this quality/speed/radius rubbish

MadMax389, 2022-12-31:
You can already do this in the printer profile...
[image]

NairoDorian, 2023-01-03:
Yes I just discovered that feature, it does help to set a min & max for the adaptative layering indeed, but not per object