bambulab/BambuStudio #904: [Bug] I can not change the chamber fan seep on windows BS

NairoDorian, 2022-12-19:
On the Windows version of Bambu Studio, I can not set the chamber fan speed.
While on the Handy Android app I can change the chamber fan speed.

When clicking any of the fan icon for the nozzle, aux or chamb, it opens a vertical window where the nozzle and aux box are on top of each other but it seems the chamber setting box would be under but it reach the bottom of the window/screen.

zeroemso, 2022-12-19:
sounds like an environment problem not a bug that can someone reproduce.

NairoDorian, 2022-12-19:
I am on a 4K display

zeroemso, 2022-12-19:
iam talkin about your windows/gfx drivers. answer the questions where you asked for on openening an issue + upload pics.
its more likely you get a response when you give as much as informations about a problem.

lanewei120, 2022-12-20:
@NairoDorian 
Please give a snapshot of your bugs
also, what is the display ratio on your windows? 100%,150%, or any other values?

lanewei120, 2022-12-20:
next time to report bug, please follow the bug format
thanks

**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

NairoDorian, 2022-12-22:
I did not go through the bug format because it appears to be happening no matter the project file.
I thought many more people had the same problems as I've seen on reddit someone had the same issue as me at least

To reproduce: just click on any of the 3 fan icons while connected to the printer, this will open a window with 2 square box on top of each other, controlling the nozzle fan and the aux fan, but the chamber control block seems to be under these 2 but I can not see it because it should be under them.

4K screen on Windows with 250% recommended scale

lanewei120, 2022-12-22:
@NairoDorian 
could you give a picture of it?

thanks

NairoDorian, 2022-12-22:
[image]

lanewei120, 2022-12-22:
@NairoDorian 
thanks,

please give a full picture of the whole studio

NairoDorian, 2022-12-22:
[image]

NairoDorian, 2023-01-03:
[image]

As you can see, the numbers to move the Z axis are also bugged.

I have the last drivers for all my hardware, latest Windows10 updated, so no issues from here

Changing the DPI scaling options via Windows helps but the Bambu Studio window is then not in 4K anymore.

walterwongbbl, 2023-01-09:
This will be fix this issue in the next version.

NairoDorian, 2023-01-20:
I have downloaded the latest 1.4.2 to ty the fix, it is working now fine with 250% recommended scaling on a 4K display, but as soon as I change to 300% or 225% or 200% I get GUI errors again, also now the extruder buttons and GUI is having problem as well.
I have included many pictures.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

walterwongbbl, 2023-01-29:
we changed the popup position of the window.  This will make the window display complete.    we will optimize the entire fan control window next.

NairoDorian, 2023-02-01:
What about the Extruder control on the right that looks broken as well ? 
Also at 300% scaling the inside control panel require vertical and horizontal scrolling to see everything, this doesn't look normal

walterwongbbl, 2023-02-02:
> What about the Extruder control on the right that looks broken as well ? Also at 300% scaling the inside control panel require vertical and horizontal scrolling to see everything, this doesn't look normal

Hi.
Because we didn't optimize studio at 300% scaling, so it will look abnormal.
We will continue to optimize in future versions.

closed 2023-08-02 by walterwongbbl