bambulab/BambuStudio #905: Allow sharing of printers in Bambu Studio with other accounts, and create an easy way to start multiple printers at once

tabiznet, 2022-12-19:
A few major things that are preventing me from using the X1 or P1P as a farm printer is the lack of ability to easily start the same print across multiple printers at once. The current situation isn't too terrible if you just have a handful of them, but I'd like to retire my existing farm, which would mean buying 50-100 printers. It would take me ages to use the drop down in the Devices section to send to multiple Bambu Lab printers at once. 

The second thing, is the ability to share printers with other Bambu Studio users. 

A few examples where this would be useful are:
My boss bought an X1 for himself that lives at the office, I bought an X1 for myself that lives at home. 
I would like for him to be able to start a print on my unit without sharing my login with him and vice versa.

A second example would be making accounts for employees that are limited to certain printers.
I have 2 Print farms, one is for production, the other smaller farm is for prototyping. I'd like to give my employees a login that's restricted to the production farm, and not the prototype farm.

**Describe the solution you'd like**
A "Send to all connected printers" button that sends the current build plate to every printer connected to the account.
Alternatively, a window that pops up with check boxes near each printer and a "Select all" feature.

A "Share printer" button where I can add the email of another Bambu Studio account and have it appear in their instance of Bambu Studio.

**Describe alternatives you've considered**
If the X1/P1P could be made to support Octprint I could use my existing farm software. Not sure how that would work considering the lack of external USB functionality.

Seb-Monier, 2023-07-09:
Hello, 
I'm also interest for that: making accounts for employees that are limited to certain printers.

wittrup, 2023-08-10:
We support this idea!
Even though we only have a small prototype farm, we already have issue with people being logged out by each other, unable to remote start and the whole thing limits prototype production.

sinhan11, 2023-08-10:
We also need this function!! 

Despite the limited size of our prototype farm, we are already encountering challenges such as users getting logged out due to overlapping sessions, difficulties with remote starts, and overall constraints on the production of prototypes.

YvoTuk, 2023-08-12:
Same here, not multiple printers, but would like to share a printer for two users.

c0ryd, 2023-08-16:
I would also like this feature.  I have access to 3 bambu printers, but they all need to be on separate accounts due to owner/location.

markinNZ, 2023-10-30:
+1 I'd like my kids to be able to print their own creations.  I don't want them playing in my account.

wittrup, 2023-11-08:
A quick fix for us, would be an option to somehow remote logout/sign off the user from the printer.

Then we could find IP by MAC address and generate sign in QR by:
```python
import qrcode
import qrcode.image.svg

ip = ""
deviceId = ""

URL = f'https://bambulab.com/en-eu/device/qr-code?deivceType=BL-P001&deviceId={deviceId}&ip={ip}&tls=1'
print(URL)

img = qrcode.make(URL, image_factory=qrcode.image.svg.SvgImage)

with open(f'qr-{deviceId}.svg', 'wb') as qr:
    img.save(qr)

```

iron-bro, 2023-12-05:
Please Bambu!!! We have a rapidly growing print farm (currently at 10 machines but soon to grow to 50) and this feature would save us so much time. 

We currently have both Creality K1 Max's and Bambu X1C's in our print farm. Creality have already nailed this need with their one-click printing solution in Creality Print

[image]

PLEASE PLEASE PLEASE build this into Bambu Studio! We actually perfer the X1C's over our K1 Max's but find it annoyingly time consuming to manage prints on the Bambu printers

walterwongbbl, 2023-12-06:
Hi.  @iron-bro 
We have indeed received many demands to manage multiple Bambu printers. We would like to collect requirements, such as which features you want.

gte619n, 2023-12-06:
@walterwongbbl 
While advanced an permissions model and fancy cancel/retry/etc stuff would be nice, I think most people would just like to be able to add one or more printers to the account, then switch between them in the UI on the app and on the desktop. Our use case would be to simply switch between the three X1s we have in our lab.

walterwongbbl, 2023-12-07:
Hi. @gte619n 
We already support binding multiple machines in one account. Do you want to send and print to these three X1s simultaneously

iron-bro, 2023-12-10:
> Hi. @iron-bro We have indeed received many demands to manage multiple Bambu printers. We would like to collect requirements, such as which features you want.

@walterwongbbl this is GREAT NEWS! I would love to help with contributing requirements and feedback. 

Right off the bat, here are some of the most important requirements I would recommend:

1) **One Click Printing**
This feature would allow users to send a single sliced file to multiple printers simultaneously. This is super important to have when run a 3D print farm business like my company does. 

For this feature to work, it should allow users to select multiple printers in one window and select an AMS slot (or slots if using multi-color's) for each printer. Once these have been selected, users click on a single button that will send the sliced prints to all of the printers they selected and will use the AMS slots that were selected for each printer. 

2) **Multi-printer status dashboard**
This feature would allow users to see the status of all printers in a single window. It would be extremely helpful for print farm businesses like us if that dashboard could also be opened as a separate window from the Bambu Studio window because we could then move that dashboard over to a second monitor and have that displayable in real-time while working in and around the print farm. 

The dashboard should allow users to access the camera feeds for each printer if needed. It should also provide a simple overview that shows the user information like how much time and percentage of time the print has left, what AMS slots are being used and the filament type/color in those slots, if there are any issues that have paused the print (maybe a red alert icon is flashing for any printers that have paused with an issue, a Thumbnail of the models being printed, etc. 

3) **Send sliced prints to multiple printers internal SD storage**
This feature would allow users to slice a model and then send the g-code to the internal storage of multiple printers with a single click (once printers have been selected) 

4) **Start prints simultaneously for multiple printers from internal SD storage**
It would be extremely helpful to have a feature where I could see the internal storage g-code files that I have previously sent to various machines and select one g-code file that is on one machine and have a button that allows me to select any other printers that are not currently being used but have this same sliced g-code file on their internal SD storage and then use the one-click printing button to get all available printers with that particular stored g-code file to start printing it simultaneously. 

I would recommend you download and have a look at how Creality Print do these things. Although I do not like using many parts of their slicer for our K1 Max's, the Multi-printer features that they have built in are incredible! and very much needed for anyone using multiple printers. We just need that same functionality with Bambu Studio.

Danitegue, 2023-12-10:
As the title of this thread points, it would be more than convenient to share one unique printer between multiple users.
The idea is that a company where one printer is used by different users (or different teams), every user (with its own bambulab studio account) must be able to send jobs to the unique printer whenever it is free.  

I would suggest to be able to select which profiles (quality profiles, and filament profiles) are shared or local.
In this way the other users could access to already calibrated filaments, or reuse profiles already created by other users.

In these shared printers, I would add a final print pop up window asking the user if the printer is ready for the next print ("Have you cleaned the bed and everything is ready for the next print?"). The answer would act as a flag that would allow other users start their print jobs.

walterwongbbl, 2023-12-12:
@iron-bro You explained it very clearly, and we will discuss these matters.

@Danitegue @gte619n 
This involves account permissions and printer permissions. I will communicate your requirements to the design team.

phifogg, 2023-12-29:
+1 ... as I am sharing a P1S with my son. We would like to be able to use the printer from 2 different bambu accounts without the need to logoff/logon again every time. Thanks!

johanroug, 2024-02-02:
hi
Need this. I am currently looking for best solution. Considering other printers because this feature is missing. We have hundreds of internal users, that should be able to print on the available printers

netechsys, 2024-02-06:
Hi.  I too would love to request this. I have multiple printers, 2 at home, 2 in the office. And would love to be able to have employees attach and print to the office printers only, and not my home printers. I am looking to add more printers in the office and building a farm for internal usage. 

This would make work flow very nice!!

bfmetcalf, 2024-04-05:
We purchased at my company an X1C and now realized we can't bind multiple users to the single printer.  This hampers us a quite a bit on our use case.  Others in the company are interested in our results and may be purchasing more in the future, but without multi-user access to a printer it's probably a deal breaker.

tbair8, 2024-04-12:
I am in the same boat as others. We have an X1 Carbon at work that needs to be shared by multiple users (100) and would love to add more printers, but we need an easier way. I also have printers at home, and I would prefer to avoid having to log into separate accounts.

M1S2, 2024-04-17:
Same problem here. My brother and I have an A1 and we want to control the printer using our own Bambu accout respectivly. Using the option to login via our Google accounts makes it impossible to share accouts.
The only way at the moment is to create a third common account with a new mail address used from both of us.
This can't be the way to go!

ciroiriarte, 2024-06-25:
seconding this!, don't have a farm to send a single job but I would like to share my X1 Carbon with my brother.

tabiznet, 2024-06-25:
The fact that this still isnt a thing is wild to me lol

pl4nkt0n, 2024-06-27:
One-up for this. In our Company we have a couple colleagues with both private and work X1C and X1E's.
at the moment we're using app twins on android phones to manage the printers.
Anyway, that' not possible on iPhones. Would be great feature for Bambu Studio and Handy :)

ziporah, 2024-07-05:
One up for this as well. Would love to be able to share my printer with my sons and daughter. In a family context I don’t want to share my account and just want to grant them access to use the machine.

timothyzalusky, 2024-07-25:
Going to throw my name in the hat for this as well

redguy212, 2024-10-03:
> @iron-bro You explained it very clearly, and we will discuss these matters.
> 
> @Danitegue @gte619n This involves account permissions and printer permissions. I will communicate your requirements to the design team.

@ChloeShang123  @lanewei120  @walterwongbbl 
Hi, Big fan of the printers and the software; adore the work you are all doing!
This feature is almost necessary for my continued, expanding use of Bambu's printers (education), and I really would love to buy way more devices.

Its been 10 months or so since you brought this to your team, has there been any progress on this?
Is there anyway the community can assist you in the development of these features?

lanewei120, 2024-10-09:
Hi guys

We have already supported multiple printers at 1.9.1 serveral months before
[image]

and for sharing printer with different accounts
it is not an issue can be done at studio side

I have forward this requirement to our device&&cloud team before,
and suggest to disccuss it on the bambulab forum

mathis-m, 2024-10-26:
I do not understand why a initial version of this for home use would need more discussion. The issue is open over a year now, which is sad to see.
No permission model or so would be needed to be added in a first iteration.
It would just be enough to allow the sharing of printers to multiple accounts.
In a second iteration this could of cause be reworked and extended to support permission models for company use.

Goose66, 2024-10-29:
+1. Need to let students in the Innovation Lab of our school as well as robotics team members have their own Bambu accounts and Bambu Labs app installed to manipulate files, but be able to send to printers (currently 2 X1Cs and an A1) over the LAN and not the current microSD card "sneakernet." The three Bambu printers all currently belong to a teacher account. Queued prints and teacher/admin approval would be a plus, but simply print to 1) Bambu printers on the LAN and/or 2) printers in someone else's account would be our initial requirement.

timothyzalusky, 2024-10-29:
Yeah, these are all features that we should have had a year ago, we aren't asking you to develop some new algorithm to make our prints better or anything crazy, we are just trying to use our printers well within groups and organizations.  These tools/capabilities should not take long to implement and at this point we are really confused why it hasn't happened.

chrisn-au, 2025-01-12:
Guys,

I am joining the long list of people requesting this feature. It is a no brainer requirement , I appreciate it may be difficult technically (and not just a studio requirement ) however it is a security nightmare to be sharing a single login  which is what is required - is there anything the community can do - if it was open source I would be happy to do a pull request

timothyzalusky, 2025-01-12:
Bambu Lab, what is the proper means to submit this request since it is not a slicer related improvement and is tied to Bambu Cloud?  Should I just start making tickets on the website until you add this feature that will be used by probably 30%+ of your users?

lanewei120, 2025-01-13:
hello @timothyzalusky ,

Since this feature is primarily implemented on the cloud and device side,
please visit the Bambu Lab forum or create a user ticket to submit this request

we can only process the feature related with bambu studio here

thanks

kiernangs, 2025-01-26:
> and for sharing printer with different accounts
> it is not an issue can be done at studio side

Can you clarify where and how to do this? I have not been able to find this ability anywhere.

alxkos, 2025-01-27:
from 2023 still nothing ((( please add multiple users per printer!!!!

Goose66, 2025-01-29:
If I am using LAN Only mode on my printers, should I be able to connect multiple Bambu Studio installations to a single printer? It would seem so, being MQTT-based and all, but I haven't been able to get it to work. It appears that one Bambu Studio installation is acting as "primary" and while others can connect, they almost immediately drop the connection. Not sure if this is a limitation in the Bambu Studio app or a limitation of the Bambu printer firmware.

maduino, 2025-10-03:
This is exactly what we need!
We have several users on two printers, and people keep logging each other out from the printers! Please add this soon!

nikhilgv9, 2026-02-03:
Really need this feature for our company to function.

c0ryd, 2026-02-03:
**Since this feature hasn't seemed to gain any traction from Bambu yet, I built a workaround for MacOS: [Bamboo Switch](https://corydawson8.gumroad.com/l/qmriaz) — a profile manager that lets you switch between Bambu Studio accounts without logging in/out each time.**

 Each profile keeps its own credentials, so you authenticate once per account, allowing you to switch freely between accounts.  **Useful for:**
 - Switching between home and work printers
 - Avoiding repeated 2FA when moving between accounts

 [Free CLI via Homebrew](https://github.com/c0ryd/bamboo-switch), or there's a [menu bar app](https://corydawson8.gumroad.com/l/qmriaz) if you prefer clicking over typing.

 Not a replacement for native printer sharing, but it makes multi-account workflows much less painful.

Skye-Bambulab, 2026-11-16:
Hello, thank you for sharing your feedback. We understand the need for multi-account printer sharing and batch task management.

We would like to share our current perspective on this:
1.Product Positioning: Bambu Studio is primarily designed as a personal creation and slicing tool. Large-scale fleet management is not within the core scope of its current architecture.

2. Recommended Solution: For users managing multiple printers,we recommend using BambuFarmManager, which is specifically engineered for multi-printer scheduling, task queuing, and professional fleet monitoring.
 
3.Future Considerations: At this time, we do not have plans to implement multi-account printer sharing, as it requires significant changes to our security and data architecture.

We understand the use case and will keep this as a feature request for future evaluation.

phifogg, 2026-11-16:
Just wanted to clarify, I am not interessted in BambuFarm... I only have one printer. But I would like to share usage with my son ideally we both have our own bambu account but still both can use the same printer. Right now we have to share one account.

Danitegue, 2026-11-16:
@Skye-Bambulab "3.Future Considerations: At this time, we do not have plans to implement multi-account printer sharing, as it requires significant changes to our security and data architecture." 
What about to implement an "allowed extra users" feature?. 
So that the printer is owned and governed by only one account, but that owner could define for a given printer, a list of other users that are allowed to send jobs to his printer. 
I.e: 
I have the printer registered with my bambulab account.
In my account settings, I could add the bambulab accounts of my wife and my son, so that they can send jobs to my printer, when not busy.