bambulab/BambuStudio #907: AMS humidity value in MQTT is broken/wrong after firmware update of X1C/AMS

cameo69, 2022-12-19:
**Describe the bug**
After updating to the latest version of X1C and AMS firmwares (FW) on 18 Dec 2022 the AMS' humidity value delivered via MQTT is no longer correct plausible. Before the FW update the values delivered ranged around 20% which matched more or less the value measured with a external hygrometer. After the FW update the values delivered are 4 and 5, which does not match the measured humidity inside the AMS at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to X1C MQTT broker
2. Subscribe to topic "print"
3. Check under print --> ams --> ams[0] --> humidity

**Expected behavior**
A plausible humidity value is expected to be sent.

**Screenshots**
Figure shows old values and then sudden change after FW update:
[image]

**Versions**
 - Model: Bambu Lab X1 Carbon
 - Version X1C: 01.03.00.00
 - Version AMS: 00.00.05.75

ashleyuk90, 2022-12-20:
This is because in the latest FW release they changed it from a % value to reporting on a scale. Bad / Okie / Good e.t.c inline with the colours shown on the Studio interface.

cameo69, 2022-12-20:
Hmmm, but why? That removes the chance to get the detailed value. Would be better to have this mapping done in the software. With this new scale, is there a mapping what the numbers represent or even better, which value ranges map to what number?

ashleyuk90, 2022-12-20:
From the changelog:

Notes: Newly installed desiccant can take hours or a night to absorb the moisture until the chamber is stable. Lower temperatures also slow down the process. Due to the sensor location, raw humidity measurement doesn't reflect the chamber state accurately. Instead, we applied an algorithm on raw data to estimate the effectiveness of the desiccants, which should be more reliable in a wider temperature range.

cameo69, 2022-12-20:
Well, I don't like it, but then it is not a bug. Closing the issue.

closed 2022-12-20 by cameo69