bambulab/BambuStudio #909: Opening camera feed crashes BambuStudio 1.4.0.17

Hercules117, 2022-12-19:
**Describe the bug**
Application opens fine. When attempting to open the device menu, opens fine. Clicking the play button to start the camera feed results in an "endless loop" of thinking, then when attempting to cancel crashes the application. 

**Expected behavior**
Should start the camera feed

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 64 bit 
 - Version 21H2

Hercules117, 2022-12-20:
Upgraded to 1.4.0.18, same issue.

Hercules117, 2022-12-20:
Issue is also in two different laptops, with two different printers, (Identical) and also on the mobile app v1.0.6.

bambu123, 2022-12-21:
We need the minidump to check the problem. 
A minidump can be generated from the windows task manager when the program is hung. 
Tell me which version of Bambu Studio are you using when sending back the minidump.

Hercules117, 2022-12-21:
The dump file was too big to upload here, 972 MB. Uploaded to dropbox, please let me know if you need me to upload it somewhere else. I'm using 1.4.0.18. 
[https://www.dropbox.com/sh/ca5gkhth8xe2w55/AABsNsuNP-uysNzsbcww78L3a?dl=0](url)

bambu123, 2022-12-21:
Ok，thank you！

Hercules117, 2022-12-21:
Also, firmware version of the x1 is 1.03.00.00

bambu123, 2022-12-22:
Not sure if it is really hung from the minidump.
I need another minidump to confirm.
Please let the not-responding error dialog show before catching the minidump.
Catch another minidump a few seconds later than the first one. Comparing the two states should get more information.

[image]

Hercules117, 2022-12-22:
Okay, I uploaded the second dump file, 10 seconds after the dialog box appeared. My computer upstairs has an older version of studio, it opened the camera just fine. (1.3.0.25) The screenshot was uploaded to the same folder. 
https://www.dropbox.com/sh/ca5gkhth8xe2w55/AABsNsuNP-uysNzsbcww78L3a?dl=0

bambu123, 2022-12-22:
Maybe this is a known issue with the dua-video-card system.
As mentioned here #244.

Hercules117, 2022-12-22:
I will say I had this issue on multiple systems, one with dual graphics cards, and one with only intel-integrated graphics. I am also having issues with the iPhone app, the video isn't loading there either after the update.

bambu123, 2022-12-22:
No, there should not be a crash dialog for the iPhone app. So are you mean the endless loading?

Hercules117, 2022-12-27:
actually, the iphone app is working now. I'm still having issues with the PC camera feed. Everything else is working. I've also moved to another laptop, same issue.

Hercules117, 2022-12-27:
*Work Around Discovered* I right forced it to use integrated graphics, but it has to be used each time for the camera feed to run. 
[image]

bambu123, 2022-12-28:
You can edit it in the NVIDIA config panel. Then the config is saved by the NVIDIA driver.

joshwigginton, 2023-01-02:
Having the same issue. Workaround seems to work ok. 
[image]

bambu123, 2023-01-03:
Yes, that's it.

closed 2023-01-11 by SaltWei

robandwend, 2023-04-29:
The nvidea control panel change worked for me.