bambulab/BambuStudio #910: Reload STL resets location and orientation

rncfiles, 2022-12-19:
**Describe the bug**
Reloading STL resets the location and orientation of the model.

**3mf File for This Bug**
Any model will suffice

**To Reproduce**
Steps to reproduce the behavior:
1. Add an object to a plate.  Move the model to a different location or change the orientation
2. Select "Reload STL" and watch the model jump back to default location

**Expected behavior**
Reloading STL should not alter how the model is set on the build plate.  For reference, PrusaSlicer behaves as expected.

**Desktop (please complete the following information):**
 - OS: Win10
 - Version 1.4.0.17

werewolfenn, 2022-12-20:
That actually sounds like it is doing just what "reload" should do. It's reverting any changes you've made and reset it back to original import.

werewolfenn, 2022-12-20:
Good point

rncfiles, 2023-02-03:
I cannot reproduce this in 1.4.2.13.   I didn't see anything related in the release notes, but happy it is working as expected now.

closed 2023-02-03 by rncfiles