bambulab/BambuStudio #911: [Bug] During Start GCode, the hotend always goes down to 140°C no matter what

NairoDorian, 2022-12-19:
I tried changing the start gcode to change this temperature of cool down before bed leveling during homing toolhead

I thought it matched the line " M104 S160 ; set temp down to heatbed acceptable" but changing that line has no effect

NairoDorian, 2022-12-22:
We have been looking into this 140C issue and people seem to find my gcode helpful https://www.reddit.com/r/BambuLab/comments/zs4cgt/fastest_start_gcode_improved_default_calibration/

SaltWei, 2022-12-22:
This is not studio issue and also not a bug.
When run G28, the printer will low the temperature to protect the bed.
If you don't want this behaviour, use "G28 T220" instead and the temperature will keep in 220 degree when run G28.

closed 2022-12-22 by SaltWei