bambulab/BambuStudio #912: Infill pattern comment interfering with wiki link

nicohmje, 2022-12-19:
**Describe the bug**
When trying to open the wiki entry for "Infill pattern" from Bambu Studio, an error pops up because the comment describing the link is missing a space attaching it to the link 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the strength tab
2. Click on "Infill pattern"

**Expected behavior**
The BambuLab wiki opens in a web browser, on to the page explaining the different Infill Patterns 

**Screenshots**

**Desktop (please complete the following information):**
 - OS: macOS
 - Version 1.4.0.17

Benderrodrigu3z, 2022-12-20:
Strange, for some reason the Sparse Infill Pattern is hyperlinked and executes with error

[image]

lanewei120, 2022-12-20:
@nicohmje 
do you mean 'Sparse infill pattern'? 
and how about other links?

could you give a snapshot of the error?

thanks

nicohmje, 2022-12-20:
> @nicohmje do you mean 'Sparse infill pattern'? 
 
Yes, apologies

> could you give a snapshot of the error?

[image]

> and how about other links? 

The other links work fine, I've not encountered this error elsewhere.

SaltWei, 2023-03-02:
fixed in 1.5.0 public beta