bambulab/BambuStudio #913: [Issue] 'Unload' button not shown when AMS is plugged in (This is a firmware issue, but there is no channel to report firmware issues on this server)

BambuLabDiscord, 2022-12-20:
After using the back spool holder on X1, the 'Unload' button doesn't show on the touch screen to unload the filament, if the AMS is still plugged in. The button will show only after AMS is unplugged.

Merlwynd, 2022-12-20:
This is the same as issue #766.....there has been no response to issue 766 yet......

Benderrodrigu3z, 2022-12-20:
heat the nozzle and hit extrude button the screen, its the Up arrow on the movement screen 
[image]

Merlwynd, 2022-12-20:
of course....or Bambu Lab could have the 'Unload" button reappear whenever the "Enable AMS" button is unchecked (meaning you are using an external spool) when sending a print.....so you don't have to do all those steps to unload the external filament....