bambulab/BambuStudio #914: [Feature Request] Print prime line in the front side of plate instead of on the back

BambuLabDiscord, 2022-12-20:
Currently an L-shaped prime line is laid down in the  back of the plate. This line is hard to remove without taking the plate off the bed. Printing a prime line in the front in the same area where the extrusion calibration lines are will make it possible to pick up everything from a textured PEI plate after the print is finished, without having to take it off the bed. This small change not only will make the plate cleanup easier but also helps to keep the ABL data reusable longer (at least for the same filament).

Benderrodrigu3z, 2022-12-20:
you can change the start gcode to accomplish this yourself, if you think its a good idea then learn the gcode like we all did and figure it out, here a hint, G1 is the motor, Y265 is spot 265 on the Y axis

[image]

Benderrodrigu3z, 2022-12-20:
it should be around the last couple lines of code in that, you could copy it all then remove the lines until that line purge line is removed and then copy them back

It might be this:
G0 Y128 E6.4
G0 X252.5
G0 Y252.5 E6.4
G0 X128 E6.4

AbeFM, 2022-12-24:
I would really like to hear what bambu has to say about this. I would like to see them consider and weigh the issue, and perhaps provide the solution to a larger audience. While a work around for a single individual makes sense, it would be great to see the product improve and the user experience improve with it.

I've always thought they should add a little tab on the second/third layers to make the end of the prime line easier to pick off the bed as well.

kibinek, 2023-11-29:
I'd strongly recommend to incorporate a new check button in Bambu Studio (e.g. in Quality setup) to let user decide whether he/she wants to print prime lines or not. Prime lines are not friendly to plate, since they are printed on the same place again and again. And if user forgets to remove them, it may cause printer damage.
I am used to use skirts for nozzle purging, prime lines are really really annoying me.

GuzziRaz, 2023-11-29:
Agreed. I assume there is a reason for not only a prime line in front (even with calibration turned off) but also one at the back, instead of just defaulting to a skirt that will never be in the way. But what it would be?

kibinek, 2023-11-29:
Well, I have solved the issue with unwanted prime lines [here](https://forum.bambulab.com/t/how-to-get-rid-of-custom-prints/38343/4?u=libor.g.vitek) :)