bambulab/BambuStudio #916: [Feature Request] sync custom filament profiles to AMS

BambuLabDiscord, 2022-12-20:
sync custom filament profiles to AMS or allow users to edit Bambu profiles and overwrite

natebluehooves, 2022-12-23:
Specifically the ability to simply add custom filament profiles to the printer/AMS and have it sync those across would be sufficient for most. 

At present, I have to avoid clicking the button to auto-populate my filament slots from the AMS in Bambu studio, and that is really unfortunate as that's a huge convenience.

jpasqua, 2022-12-23:
This would make life much better for users, especially given that we cannot add RFIDs to non-BL spools.

gitjpk, 2023-01-23:
Definitely a need for next version :)

Ali229, 2023-02-15:
Filament not synching to the ams makes the sync button on bambu slicer useless for anyone using non bambu filaments.

fadnincx, 2023-05-05:
Any updates on this front from Bambulab? I would really like to have this possibility, as I'm often printing with "unknown" materials.

eRacoon42, 2023-05-14:
+1
Most important feature request for me.

aladdin224, 2023-07-08:
+1 for this! Would love it!