bambulab/BambuStudio #917: Slicer: Split very large Objects into printable parts and distribute them among Plates

theogott, 2022-12-20:
**Is your feature request related to a problem? Please describe.**
Printing large Objects lets say 0,5 m x 0,5m can currently not be done because these Objects will not fit onto the Printer-Plate.
The Slicer can overcome this issue.
Using a recursive Algo, large Objects can be split into smaller Objects that can later be glued together to get the final Object.

**Describe the solution you'd like**
Take a look at this Sample: 
[Large Dragon in 4 Pieces](https://www.printables.com/model/316851-very-large-chinese-dragon-in-4-parts-glues-togethe)

**Describe alternatives you've considered**
Currently I have to load the Object into Blender and use the Difference Modifier to print such large Objects, see Sample: 
https://www.printables.com/model/316851-very-large-chinese-dragon-in-4-parts-glues-togethe

**Additional context**
[image]

AbeFM, 2022-12-24:
It doesn't ruin what you're saying, but a point I always have doing this is needing features to align the parts later.

It's not *critical*, but parts with alignment pegs/holes/features are a lot easier to deal with and can come out great. Certainly when I'm selling what I'm printing it's nice to have it perfect.

This is a pretty bad job I did on a recent part:
[image]

I would have liked to be able to just click "split". How do you deal with assembly when you cut with flat planes?

theogott, 2022-12-24:
I completely agree with you. So lets add your suggestion to the Feature-Wish list.

miikememe, 2024-06-06:
Would love to see something like this come to fruition!

esjortiz, 2024-09-23:
This would be great.

aanast01, 2025-05-26:
This feature request is exactly what I am looking for as well. Thumbs up!