bambulab/BambuStudio #920: Nozzle did not cool off during heatbed leveling, 1st time using 3rd party textured PEI plate.

finsdotsurf, 2022-12-21:
**Describe the bug**
This may be a bug that happens when the plate location marker is not found. For the first time using this machine I noticed the hot end did not cool down during the bed leveling sequence. I only noticed this after dismissing the "bed location marker not found" dialog. This is the first time I've used a textured plate (not Bambu's, no markings).

**To Reproduce**
Steps to reproduce the behavior:
1. Slice with textured bed chosen.
2. Cover the bed's location marker or use an unlabled bed.
3. Print with bed leveling and flow calibration.
4. After dismissing the "bed location marker not found" error and continuing the print, check to see if the hotend does not cool down for the rest of the bed leveling sequence. Should see a fair amount of oozing if this happens.

**Expected behavior**
I expected the hotend to cool down for the bed leveling sequence like it always does.

**Desktop (please complete the following information):**
 - OS: [Mac]
 - Version [1.4.0.17]

SaltWei, 2022-12-26:
This is not studio issue. Please move to https://forum.bambulab.com/ for better support. Thanks!

finsdotsurf, 2023-01-05:
Solved: In Bambu Studio's Device View, the Print Options have a checkbox for 'Enable detection of build plate position.' I unchecked it.

closed 2023-01-05 by SaltWei