bambulab/BambuStudio #921: Localization marker pause won't clear and resume from studio

randyre, 2022-12-21:
**Describe the bug**
Error about localization marker comes up but when cleared on Studio it remains on the printer and won't continue without direct user interaction with the touchscreen.

**To Reproduce**
Steps to reproduce the behavior:
1. Start print with generic build plate without markers.
2. At error clear error on Studio
3. Print will not resume
4. Cle3ar error on printer interface
5. Print now resumes.

**Expected behavior**
Error message resolution should not require a user to be physically onsite with the printer and clear the same message that was cleared in Studio.

 - Version 1.4.0.18

randyre, 2022-12-21:
This is also along the same lines as #509 

Please upgrade the priority of this and 509. This has a major impact on using the slicer and managing a printer remotely.

SR-G, 2022-12-21:
Same issue for me with latest firmware / Bambu Lab studio versions - with only one difference, i'm able to "remotely" resume the print (from Bambu Studio on Windows), but i do NOT want the print to be "paused" in case of a 3rd party plate !

This really is not going in the proper direction and is here a "regression".

I hope previous behavior will quickly be restored (or at least an extra option allowing to automatically ignore that specific kind of "warnings").

StoneLiBambulab, 2023-02-06:
The printer is not supported it yet, it will be planned as soon as posibble when the printer firmware is supported.

Thanks!

Stone Li