bambulab/BambuStudio #926: Not mapping AMS filaments if manually added filament change

leofaoro, 2022-12-22:
I wanted to print the bottom of an object in one color, and the top in another. I created a "change filament" in the "preview" tab by right clicking on the right bar and creating a "change filament" command. Once i re-slice and click send, the next window only shows 1 color, not 2 so I can map to the right spool in AMS. I sent it to print, but after it prints the first color it gives an error and I had to cancel. This is with a P1P on BLS 1.4.017 and firmware 01.01.00.00 and AMS 00.00.05.75.

debren27, 2023-01-12:
I also just ran into this. In my case, I changed only the last/top two layers. In the attached screenshot, in the top right filament usage box, you can see that it's going to use both filaments. 

Studio 1.4.0.18, X1C 1.03, AMS 00.00.05.75. I'm pretty sure this used to work.

[image]

debren27, 2023-01-12:
Digging further, this looks like a duplicate of #843, which is getting traction.

StoneLiBambulab, 2023-01-16:
Hi @leofaoro ,
     Hi, I fixed the issue and the hotfix version will be published this week.

Thanks!

Stone Li

closed 2024-01-15 by SaltWei