bambulab/BambuStudio #928: AMS wont save changes on Filament type

Quameda, 2022-12-22:
If you change the AMS slots to self generated Material and restart Bambulab studio the changes wil not be saved. At restart the slots are all Generic.
V 1.4.0.18 OSX

bambu123, 2023-01-05:
What is self-generated Material? Currently, only the system filament preset can be set to the AMS tray. Also, you can't edit the information on the Bambu Office filament.

Quameda, 2023-01-05:
Hi,

wasn’t able to send images would not let me upload. So here they are.

Setting these in the studio:

￼

then after restart of the studio it looks like this so it is NOT saved:

￼

So I have to klick and chose 8 at the moment and soon 16 !!!! Colors each and every time or I load a dummy project where the filament is saved. On the Pribter I can not choose and set my eSUN presets

This is not nice and not acceptable to be honest.

kind regards,

Peter Balogh

bambu123, 2023-01-05:
Can't see the pictures, are you mean the filament-type setting here?
[image]

Quameda, 2023-01-05:
yes if you set a manually edited and saved filament type and you reopen the Studio the setting is gone. Should not be hard to save those.

kind regards,

Peter

bambu123, 2023-01-05:
But manually edited and saved filament type (it's really preset) can't go to the Combobox here.
The list only has system presets.
[image]

If you set 'Bambu PLA Basic' here, is it saved?

Quameda, 2023-01-05:
only Bambupresets are saved. If they are edited they are not which is really annoying

bambu123, 2023-01-05:
The modification to the system preset can not be saved to the AMS tray. It is only the preset id been saved to the AMS tray.
We plan to support user preset here, but currently, only system preset can be selected.

Quameda, 2023-01-05:
if you can save it in a project, dummy project is workaround right now, you can save it in Studio, no?

bambu123, 2023-01-05:
We can save the list of filament with the preset (type) and color to project. But it is not related to AMS. So do you mean the filament list is not saved by Studio? [image]

Quameda, 2023-01-05:
why cant you see images in the email? Would be easier to explain.

bambu123, 2023-01-05:
You should attach image from github web page other than from email.

Add comment in this link.
https://github.com/bambulab/BambuStudio/issues/928

bambu123, 2023-01-13:
@Quameda Please help continue to provide information, especially screenshots of the program.

Quameda, 2023-01-13:
This is when you set and leave Bambustudio
[image]

This is when you start it again

[image]

bambu123, 2023-01-13:
I know this is a known issue with custom user presets that are not loaded at the very beginning of the program startup.
We will release a new version soon which fix this issue.

Quameda, 2023-01-13:
Ok great will you also synchronize it back to the AMS? Would be nice to set everything on PC and have one setting in the Studio and when you stand at the Printer.

bambu123, 2023-01-13:
Sorry, it is not related to AMS. To modify AMS filament info, please go to the device panel.  Also. custom filament preset can't be written to the AMS tray until the last version.

Quameda, 2023-01-13:
That’s the problem AMS and Studio Settings are related of course and should have the same settings and communicate both ways. Why should there be another setting on the AMS than in studio?