bambulab/BambuStudio #929: Maintenance managing

MarcoJPereira, 2022-12-22:
**Is your feature request related to a problem? Please describe.**
Maintenance on the X1 carbon is important and as a huge impact on the prints quality. In a 3d printing farm, listing all the procedures that have been done for each machine and when the next maintenance need to be done can be challenging.

**Describe the solution you'd like**
The wiki is already a good start to troubleshooting the X1 carbon, and even if there is a section for maintenance, I look at it only when I have time between prints. Therefore I'm not planning maintenance regularly. It would be great to have a tab in the section "Device" where we could list all the maintenance that can be done. Each item on the list could have the date of the previous operation, the frequency of maintenance (advertised by Bambulab in the wiki) and the date of the next maintenance. The line could be green when the maintenance is valid, orange when it arrives at the maintenance and red when it is invalid. In addition, a stock/shopping list could be added, like this each time a piece of the printer is changed, the stock list could be updated as well as the shopping list.

This could improve the workflow for managing the printing farm as well as tracking in the log where the problem can come from if the maintenance is logged in the printer.

AbeFM, 2022-12-24:
+1 on that one. I wouldn't call this a farm at my house, but I have enough machines much of my maintenance happens when I'm told to or I notice something wrong.

mainmind83, 2023-01-25:
As if it were the maintenance of your car. You should take into account the number of hours of operation according to the type of filament and "kms" traveled on the axes, to suggest, in addition to the information collected in the wiki, a possible change of nozzle, for example. It should be optional and configurable.

Ali229, 2023-01-26:
This could potentially reduce the number of tickets bambu receives and teach new people when and how to maintain their machines +1