bambulab/BambuStudio #934: Application lags for multiple seconds when window is dragged

akbyrd, 2022-12-22:
**Describe the bug**
The application lags when dragging the window around from the title bar.

The application appears to be doing non-trivial work while it's being dragged. With a high polling rate mouse the application can't keep up and begins to lag behind the mouse cursor. This isn't terribly noticeable at 500 Hz or below, but with a 1000 Hz mouse it's very apparent. When I stop dragging the window it can take multiple seconds for it to catch up. It's more apparent with small windows that fit entirely on-screen. It doesn't matter which tab I'm on (Prepare, Preview, etc). I see the main thread maxing out one CPU core, but I'm not entirely sure what it's doing. The top few stack frames when profiling aren't particularly illuminating.

I'm assuming either the main thread is repainting the window while dragging, the window is processing a single message per frame, or the message loop is doing something heavy for each received message.

[image]

**3mf File for This Bug**
N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu Studio
2. Click on the title bar
3. Drag the window around
4. Notice the delay

**Expected behavior**
The application should be able to paint fast enough to 

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Windows 10 22H2
 - Bambu Studio: 1.4.0.18

bambu123, 2023-01-10:
I have reproduced it on one computer. 
It may relate to the video card or wxWidgets (The UI framework BambuStudio & Prusa use which all have this problem on the specific computer).
An update to the windows system may solve it.

bambu123, 2023-02-09:
@akbyrd Have you beat the problem? 
It disappears on my computer. I do nothing but reboot the computer.

akbyrd, 2023-02-10:
The issue is still present after rebooting, updating windows, and updating the application.
Windows 22H2 19045.2546
Bambu Studio 1.4.2.13

fti7, 2023-04-16:
I have similiar issues
Windows 10 22H2 19045.2846
Bambu Studio 1.5.0.61

When Bambu Studio is open, Games like Valorant are lagging (FPS Drops, Mouse "behind" Lags etc)

Iarann, 2023-09-05:
New NVidia driver, clean boot of system, Win10Pro 22H2, no other user programs running. Bambu Studio still shows very odd behavior. Very slow to drag, continues to move after the mouse stops.

Tekunalogy, 2023-11-29:
Having the same issue as described above

Xboxer, 2023-11-29:
Same Issue
1.7.7.89 and 1.8.0.62

Windows 11 23H2 22631.2715
Ryzen 7800x3D
ASUS ROG Strix x670E-E
Nvidia 2070 Super

Fresh installed System with latest Chipset and Nvidia Drivers

RastonWarrior, 2023-12-13:
Note, this is with the latest drivers on a bare-metal reinstall - any progress on this?  Is it an NVidia driver issue or what?

ansonl, 2024-02-03:
I have the same dragging BambuStudio window inertia issue on Windows 10 with BambuStudio version 1.8.2.56.
Prusaslicer does not exhibit the slow dragging inertia issue. 

The CPU and GPU usage shown in task manager does not increase significantly when dragging the window around.

Nuc1eoN, 2024-02-23:
> I have the same dragging BambuStudio window inertia issue on Windows 10 with BambuStudio version 1.8.2.56.
> Prusaslicer does not exhibit the slow dragging inertia issue.
> 
> The CPU and GPU usage shown in task manager does not increase significantly when dragging the window around.

@ansonl The workaround is to set your mouse polling rate in the mouse software to the lowest value.

EricDimitri, 2024-03-09:
Just a question here, how many of the ones that have the issue (I am one of those)  have Nvidia  Graphic Cards?

ansonl, 2024-03-09:
I have the lagging with a 3060ti.

EricDimitri, 2024-03-09:
I don't know if it night be related to nvidia software or drivers

fti7, 2024-03-09:
> @ansonl The workaround is to set your mouse polling rate in the mouse software to the lowest value.

Didnt fix the Issue for me (Logitech G502 X; Set to 125hz)
I have an Nvidia RTX 2060 Super. (Drivers are up2date)

ansonl, 2024-03-09:
@RastonWarrior I think you mixed my post up with @Nuc1eoN.
But anyways I did try setting the mouse polling rate lower and still had the lag.

Nuc1eoN, 2024-03-09:
How do you asses it is only an Nvidia bug?

I'm on AMD but lowering my polling rate fixes it.

You are the first people I encounter for which it would not be related to the mouse polling rate.

EricDimitri, 2024-03-09:
I don't know, just trying to find a pattern, because I had some sort of
similar issue in the past with other app, and was related to nvidia... But
it is just a guess, trying to figure it out

RastonWarrior, 2024-03-09:
> @RastonWarrior I think you mixed my post up with @Nuc1eoN. But anyways I did try setting the mouse polling rate lower and still had the lag.

I apologize @ansonl , I sure did - corrected.

Confirmed, not related to mouse polling rate.  It's possible however that it's set within the system differently: I use a Corsair mouse with iCue software that determines the polling rate, not Windows-native.  You're also saying you're on AMD and you did have the problem - okay, so not NVidia only, good to know.

Nuc1eoN, 2024-03-09:
So between the highest and lowest possible polling rate there is no perceivable difference while dragging the window for you?

Because I've noticed that it is not an "on" or "off" type of bug. Rather at my lowest mouse polling rate it is so smooth that I would not call it a bug anymore. But at 1000hz it is unusable. So at 125hz there is still an underlying issue for me, but so vague that I don't care about it so much.

Would be nice if you could verify, whether there is ANY DIFFERENCE between 125hz and 1000hz polling while dragging the window? Thanks.

PS: I also have a corsair mouse, but pre icue.

RastonWarrior, 2024-03-09:
> So between the highest and lowest possible polling rate there is no perceivable difference while dragging the window for you?
> 
> Because I've noticed that it is not an "on" or "off" type of bug. Rather at my lowest mouse polling rate it is so smooth that I would not call it a bug anymore. But at 1000hz it is unusable. So at 125hz there is still an underlying issue for me, but so vague that I don't care about it so much.
> 
> Would be nice if you could verify, whether there is ANY DIFFERENCE between 125hz and 1000hz polling while dragging the window? Thanks.
> 
> PS: I also have a corsair mouse, but pre icue. I just change it in the mouse software aswell.

Confirmed.  Clearly, if you lower the resolution of the mouse causing the pointer to move more slowly, when you drag the Studio window around it moves around even slower, but the relationship of mouse speed to window lag is always 1:1 .

Nuc1eoN, 2024-03-09:
@RastonWarrior I think you are misunderstanding. We are not talking about mouse pointer speed, but polling rate. Those are to different things. Polling rate is the frequency at which the mouse communicates to the usb port of the computer. Usually this is 125hz, but gaming mice often go up to 1000hz or even higher. It is not about mouse pointer speed but rather response time of the mouse (however you are unlikely to even perceive a difference.

However in this bambulab bug, it seems to make a difference for lots of people.

RastonWarrior, 2024-03-09:
I do not misunderstand - whether dot-pitch, frequency, pointer speed, precision, brand, driver or phase of the moon, messing with the mouse has no impact on Bambu Studio window-drag latency, as also affirmed by @fti7 and others.

Nuc1eoN, 2024-03-09:
@RastonWarrior Ok maybe Nvidia is a special case then.

But for AMD folks lowering the mouse polling rate makes OrcaSlicer usable.

RastonWarrior, 2024-03-09:
> @RastonWarrior Ok maybe Nvidia is a special case then.
> 
> But for AMD folks lowering the mouse polling rate makes OrcaSlicer usable.

Okay, can't attest to that.
I paint what I see,

  - R

EricDimitri, 2024-03-10:
Guys, I dont know if this helps, but I noticed that could be related, I think, to page faults. Thousands of pagefaults per second, even when the Bambu Studio is not touched for minutes. Page faults keep growing and growing, and that looks weird. And again, at least 2,000 (two thousand) page faults per second. That is not ok.

fti7, 2024-03-10:
Ok so i tried to replicate it better. I can see some improvements with 1000hz vs 125hz performance while drag&drop the window.
On 1000hz after some seconds, the window movement start lagging. With 125hz not.
Looks like some interrupt queue or similiar is running full.

But when i play for e.g. Valorant, i can feel a permant "lagging"

I can also confirm the ~2000 page faults per second. But other Processes do more.

EricDimitri, 2024-03-10:
DO you have any process that, while doing " nothing"  has more than 2000
faults /sec?

Other detail is that It stays on zero, or almost zero (normal) page faults
if you are in the Prepare tab, but it skyrockets if you go to any of the
other tabs.

Just checked page faults, and if I am not doing anything on the computer,
no page faults at all on the other apps. or in the order of 10s, not 2 or 3
thousands at least.

Reimajo, 2024-03-29:
This is so crazy, I just bought a 1'500€ printer and they can't fix such a basic issue with their software in 1.5 years? Really gives me the wrong first impression here, because the print itself is amazing. But using this software feels so wrong.

Just pause the window redraw while it is being dragged if you don't want to fix the underlying issue, it's really not that hard. Two lines of code. Or limit the redraw while being dragged to 60 times per second, for example, this would be the proper solution. 

For example, you can use WM_SETREDRAW which is part of the Windows user32 DLL and is described here: https://stackoverflow.com/questions/587508/suspend-redraw-of-windows-form

bzip22, 2024-05-03:
another vouch for poor window responsiveness when using a mouse with a poll rate of 1000 with it disappearing at 250.

KorbenDev, 2024-05-17:
Same problem here. Fresh install of Bambu Studio 19.1.66 on Windows 10 (10.0.19045 Build 19045).
Hardware ist a HP ProDesk 600 G2 DM (Intel Core i5-6500T, using integrated graphics).

KorbenDev, 2024-05-17:
Since this is based on PrusaSlicer I tested it (current version 2.7.4): it seems PrusaSlicer also has the same issue, but by far not as noticeable and annoying as Bambu Studio has... so something makes it worse.

Nuc1eoN, 2024-05-17:
@KorbenDev did you try to lower your mouse polling rate (if you have such possibilty with your mouse). If yes did it help?

KorbenDev, 2024-05-17:
@Nuc1eoN  No I didn't. And it sounds like a workaround to me, not like a fix.

I also tried Bambu Studio on my main computer (AMD 5950X, 32GB RAM, RTX 3080, Windows 11 23H2) - the application works fine there, at least I couldn't detect any lag. But I guess that's only because the hardware does compensate for the flaw in the application...

RastonWarrior, 2024-05-17:
@Nuc1eoN Mouse polling rate has been discussed before and indicates a sharp misunderstanding of the problem, which is that lower the polling rate causes the window to lag slower... because you're dragging it around the screen slower.  The lag behind the mouse, however, remains consistent no matter what the polling speed is (if you set the polling rate to 5, now everything is slow - that doesn't make it "problem solved".  If this workaround helps anybody, ever, under any circumstance, then they're not experiencing this problem or they're referring to other software like OrcaSlicer... not Bambu Studio.

@KorbenDev That's the odd thing, hardware doesn't accommodate per se (which is to say a better graphics card eliminates the problem) - it seems to happen with some cards and drivers, and not on others.  You have a great point about trying Prusa too and seeing what it does, I didn't check that.  Since it's still there just not as bad with Prusa than on Bambu Studio tells me it's something in the back-end code that's exacerbated by something Bambu put in there...?

fti7, 2024-05-28:
This is definitely some kind of IRQ Overload/Exhaustion

There is a way to to reduce Impact to the GPU. Together with Mouse Polling Rate to 100hz, i can at least play some Games with BambuStudio opened.

Changing the GPU Mode from Line-Signale Interrupts to Message Signaled (MSI) helps me.
https://www.reddit.com/r/OptimizedGaming/comments/107blhi/msi_mode_on_gpus/
But it seems to be standard for newer GPUs like the RTX 3000 Series. (I have an RTX 2060)

RastonWarrior, 2024-05-28:
> This is definitely some kind of IRQ Overload/Exhaustion
>
> There is a way to to reduce Impact to the GPU. Together with Mouse Polling Rate to 100hz, i can at least play some Games with BambuStudio [...]

Not sure we're talking about the same issue - I can play all games without exception (and that's a lot...) while the app is up and printing. If I put the game in a window and move that window around while it's not paused, it just moves... no big deal, but the Bambu studio app still lags like always.  Gameplay doesn't matter, nor mouse polling rate or other GPU settings.

fti7, 2024-05-29:
> Not sure we're talking about the same issue - I can play all games without exception (and that's a lot...) while the app is up and printing. If I put the game in a window and move that window around [...]

Guess your GPU is already in MSI Mode/Newer.
It has the same root cause at the end.

I will upgrade my System to Win11 next Week. Lets see if this issue is still there then (Same GPU)

fti7, 2024-06-02:
FYI: I've switched to Windows 11 Pro + new CPU/Mainboard. GPU still the same
BambuStudio Lag Problems are gone. Even when Mouse is on 1000hz

Duglum, 2024-08-26:
Can confirm that the window moves extremely slow and sluggish, it's the Mouse polling rate. 1000hz is super slow and sluggish, 125hz works as expected. Orca Slicer has the same issue while Prusa Slicer works perfectly fine.

Win10 22H2 19045.4780
Orca Slicer v2.2.0-dev nightly from today.
AMD 7950X3D, Nvidia RTX 4090. All drivers are up to date.

This is extremely annoying, hope it will get fixed soon?

malventano, 2024-11-15:
Confirmed with 1.10.0.89 on Windows 10 + RTX 3090. This is the only app I've found to be so unresponsive while dragging.

EricDimitri, 2024-11-15:
FYI, my problems were solved after the new Release of "Nvidia App" (that is called now). I did a clean install, and propblems solved. I dont know if that will be the case for you, but at least give it a try.

malventano, 2024-11-22:
> FYI, my problems were solved after the new Release of "Nvidia App" (that is called now). I did a clean install, and propblems solved. I dont know if that will be the case for you, but at least give it a try.

What "Nvidia App"?

malventano, 2024-11-22:
> FYI, my problems were solved after the new Release of "Nvidia App" (that is called now). I did a clean install, and propblems solved. I dont know if that will be the case for you, but at least give it a try.

I don't believe Geforce Experience / Nvidia App has a profile for Bamdu Studio, so that should not have caused any difference.

EricDimitri, 2024-11-22:
> I don't believe Geforce Experience / Nvidia App has a profile for Bamdu Studio, so that should not have caused any difference.

What I think is, may be they recoded drivers with that? may be some settings of the old app that were disabled were the problem? I think It is related to Nvidia, but not sure in which way, but system wide, and not specifically to Bambu

malventano, 2024-11-23:
> What I think is, may be they recoded drivers with that?

Updated to most recent Nvidia drivers on two separate systems here. Issue remains. Only Bambu Studio and Orca Slicer have this issue. Also noted that Bambu Studio pegs 4 CPU threads while the window is being dragged around (even on the startup page, so no model rendering is taking place), so some code is doing weird things when it shouldn't.

Duglum, 2024-11-23:
Updating the nVidia drivers to 566.14 didn't change anything.