bambulab/BambuStudio #935: Fuzzy skin "Inside Only"

PLAmenPRINT, 2022-12-22:
I am printing cone objects (varieties of speaker waveguides) and for more aesthetically surface I am using fuzzy skin.
The point is that there is no need to have fuzzy surface outside the object and it will be great to have a way to skip that effect than even make the printing time longer.
[image]

Could You please implement option "Fuzzy skin Only Inside"?
Thank You for the attention
Best Regards

AbeFM, 2022-12-24:
Possibly you could be allowed to paint on fuzzy skin like you do colors with the AMS?

PLAmenPRINT, 2022-12-27:
> Possibly you could be allowed to paint on fuzzy skin like you do colors with the AMS?

That will be very good solution but there isn't (or I couldn't find) an option to use "fuzzy" effect as a color.

AbeFM, 2022-12-27:
I doubt it's there, but I think it might be a good way to implement it. A suggestion for the Bambu team and to encourage further discussion.

suit, 2023-02-01:
if printed mod plates for the P1P recently with a fuzzy skin - took loads of time, and if i could have disabled it for the "inside", this would have been very useful - i support this feature request ;)

castingflame, 2023-11-12:
Paint on Fuzzy please!

exzenter, 2023-11-16:
push! all the workarounds are not perfect.

berkeozkir, 2023-11-22:
I am also trying to do outside only fuzzy skin on an electronics enclosure and have the problems you described. Is there a way that we could code an add-on or extension working with Bambu studio. Or if I could modify the original 3d model itself to have fuzzy skin in the step file, Would that print just as same?

magnusmyklebost, 2024-11-27:
implemented on prusa. You could have been first with the smart feature, but youve waited two years. Please fix! https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1