bambulab/BambuStudio #937: Windows runtime errors - fresh install 0f 1.40

WZ9V, 2022-12-23:
I got a new laptop with a fresh Windows 11 install and I get the following errors on startup when running BambuStudio.

[image]

[image]

[image]

rjdaggett, 2022-12-25:
Same here, trying to get it running for the first time. not a sign of confidence. I opened a ticket with them too

rjdaggett, 2022-12-25:
https://learn.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170

Install visual c latest rebott and it works