bambulab/BambuStudio #938: Feature - menu option to run flow calibration only

WZ9V, 2022-12-23:
Rather than run a flow calibration per print, I would like to be able to run a flow calibration and have it display the values on the LCD so that I can plug them into the filament setting in BambuStudio.  Even better would be a BambuStudio feature to run the calibration and save to the current filament automatically.

AbeFM, 2022-12-24:
I've asked this before.
One thought I had was an override (1.2x measured value, for instance)... But I would like it stored with the filament profile.

A good reason for this (I feel) is when you have two of the same material in different colors from the same supplier, but one won't read with the lidar. Then you could load a roll of, say, blue and let the machine calibrate, and then apply that value to your roll of (let's face it) black.

Also, for printing very soft materials I'm not always convinced the values are quite right - thought until I can tweak them I'm only guessing.

NairoDorian, 2023-01-10:
Yes yes yes
Transparent materials and textured bed do not allow good flow calibration with lidar, so let's use the calibration of the other filament of the same material and supplier, make the calibration data of the lidar visible and manageable, please