bambulab/BambuStudio #939: Bambu Studio warns me about overhangs but then adds no supports when I enable them

c2h5oh, 2022-12-23:
**Describe the bug**
For certain models if I slice them with supports disabled I get a warning about large overhangs suggesting I enable supports. If I enable supports and slice again the warning disappears, but no supports are added

**3mf File for This Bug**
[18650_x_3_battery_holder.zip](https://github.com/bambulab/BambuStudio/files/10295594/18650_x_3_battery_holder.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Add attached 3mf file to plate
2. Slice with supports off, note the warning and amounts of filament used for different things in the print
3. Enable supports
4. Slice again, note the warning is gone and amounts of filament used for different things remain unchanged. there is no difference in preview either.

Printing on textured plate with PETG using Generic PETG profile if that matters, but I'm 95% sure I've seen this happen with PLA on cold plate too.

**Expected behavior**
Supports should only be recommended to me if slicer is actually going to add supports

**Desktop (please complete the following information):**
 - Windows 10, Linux
 - Bambu Studio v01.04.00.17

tangketan, 2023-01-09:
Thanks for reporting the bug. We'll fix it in the next release.

c2h5oh, 2023-01-21:
This is still happening in 1.4.2.13

tangketan, 2023-02-01:
@c2h5oh Sorry, the fix is not included in 1.4.2.13. Please wait for the next version ...

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei

fishcake919, 2024-08-26:
Hey everyone, I hope you're doing well. It seems like the issue still hasn't been resolved. Just to provide some further information, it seems that the occurrence of the issue is affected by the layer height of the print. For instance, when the layer height is set to 0.2, the issue doesn't occur, but when it's set to 0.1, it does.