bambulab/BambuStudio #940: variables first_layer_print_min[0] with wrong values on second print plate

hliebscher, 2022-12-23:
variables first_layer_print_min[0] with wrong values if not the first print plate

I use M557 (G29A) for define Mesh Area

M557 X{first_layer_print_min[0]}:{first_layer_print_max[0]} Y{first_layer_print_min[1]}:{first_layer_print_max[1]} S25

if i use a second build plate i get wrong values in GCODE back

M557 X804.147:854.147 Y-531.917:-481.917 S25

;===== bed leveling ==================================
M1002 judge_flag g29_before_print_flag
M622 J1

M1002 gcode_claim_action : 1
**G29 A X353.451 Y85.9557 I50 J50**.         <<<<<<<< x max 256!!!!
M400
M500 ; save cali data
M623
;===== bed leveling end ================================

the slicer take x0 y0 from the first plate

SaltWei, 2023-01-03:
We have fixed it in internal version and will be included in next release.
Thank you very much for reporting issue to keep software improved!

QingZhangBambu, 2023-03-02:
We've fixed it. you can get that in version 1.5

closed 2023-03-02 by QingZhangBambu