bambulab/BambuStudio #941: [Feature Request] Allow users to set Z offset

BambuLabDiscord, 2022-12-24:
Add a feature to allow user to set Z offset. The Z offset can have a checkbox (by default it is checked) that uses the default/automatic). But once a user un-checks that box, users should be able to enter a custom Z offset value. It will be better if the default Z offset was shown (but grey out), as it can help users to decide how much they may need to lower or raise the Z offset from the default value. This feature should also have a boundary so that if a user enters a value that can damage the bed, it will not accept it.

zeroemso, 2022-12-24:
serious question. For what reason you need a different offset?

AbeFM, 2022-12-24:
I've had the same wish from time to time, but frankly, I've always gotten it to work after finding another problem - so I'm curious as Zeroemso is: What's the reason? I generally like more manual control and I'm curious to hear what you're going for/problem you're having.

RyanEwen, 2023-01-22:
On my last printer (Prusa MK3S+) I used different z-offsets for PETG than for PLA. Curious how this is going to work on my P1P...

suit, 2023-02-01:
you can set the z-offset in the initial start gcode, it is appearantly set to -0.04 für the PEI sheet to avoid scratching it

i've figured, on some filaments (for example black matte PLA it just terribly adheres to the build plate, manually changing it to -0.03 resolves this issue - so a manual z-offset correction (per filament or build plate type) would be beneficial

SaltWei, 2023-03-02:
Z offset can be changed in custom gcode if you want. Just like start-gcode.
[image]
After evaluate we decide not to add this option on slicer. 
1 We assume the z offset should be handled by printer itself. Since in the future, model creator may share there 3mf to other people. Z offset when slicing make the gcode and 3MF uncommon.
2 Newbie of 3D printer may destory the printer by accidentally entering the wrong value.
3 This option make 3D print more hard to use. Bed level and z offset should absolutely handled by printer itself.

If have any problem with Bambu printer about z offset and bed level, please report issue to us.
Thank you very much!

closed 2023-03-02 by SaltWei