bambulab/BambuStudio #945: Option to change solid infill pattern

djngnear, 2022-12-24:
It would be great to port over Superslicer's solid infill pattern settings option. When I select Monotonic for the bottom and Monotonic for the top, the solid infill pattern is Monotonic. However, when I change the top pattern to Monotonic Linear, the solid infill pattern changes to Rectilinear. And that sometimes leaves a poor quality top layer.

[image]

SaltWei, 2022-12-29:
> And that sometimes leaves a poor quality top layer

Could you share the quality picture for us to debug?

SaltWei, 2023-06-20:
Same with https://github.com/bambulab/BambuStudio/issues/1019.
Will be included in next release.

SaltWei, 2023-07-11:
Included in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.53
Thank you very much for suggestions!
[image]

closed 2023-07-11 by SaltWei