bambulab/BambuStudio #949: [Enhancement] exclude selectable AMS Slots from "Flush into objects' infill"

DasKashi, 2022-12-25:
**Is your feature request related to a problem? Please describe.**
Biggest problem: can't or at least shouldn't use this option with support filament as this would vastly diminish the printed parts' structural integrity

**Describe the solution you'd like**
selection for AMS slots not to be flushed into objects' infill when that option is active

**Describe alternatives you've considered**
currently, when using support filament, I just wont flush into infill

**Additional context**
example of Support W being flushed into infill
[image]

randyre, 2022-12-26:
This should really be labeled as a bug. Support filament as infill as a very bad idea and it should never be allowed to happen, but the other filaments need to still flush like normal. Just unchecking the box isn't really a viable option.

DasKashi, 2023-01-30:
Duplicate of #1207 

reopened as Bug