bambulab/BambuStudio #950: When switching to Previous tab with Ctrl+Shift+Tab, two actions are performed

KD7LRJ, 2022-12-25:
Using the hot key Ctrl+Shift+Tab to switch to the previous tab (e.g. from Preview to Prepare) cause two actions to be performed instead of just the desired action. Moving to the next tab with Ctrl+Tab works fine. Alternate shortcut keys of Ctrl+PgUp/PgDn also perform the Previous/Next tab action correctly. Note that none of these very handy keyboard shortcuts are listed in the "Keyboard Shortcuts" popup.

Steps to reproduce the behavior:
1. Have any model loaded and be on the Preview tab.
2. Move to the previous tab by pressing Ctrl+Shift+Tab.
3. The view will switch (correctly) to the Prepare tab, but will also perform the action of "Collapse/Expand the sidebar" each time you move to the previous tab.

**Expected behavior**
ONLY the single action (move to previous tab) should be performed. The Shift+Tab action (which is also a part of this key combination) should not be performed.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

lanewei120, 2023-01-11:
thanks for reporting this issue

these two shortcuts keys are similiar
we will solve the conflicts soon

lanewei120, 2023-01-31:
already fixed in v1.4.2

closed 2023-01-31 by lanewei120