bambulab/BambuStudio #952: [Issue] Bambu Slicing Bug (layers just get deletet in slicing and will not be printed).

BambuLabDiscord, 2022-12-26:
Single model Sliced in Prusa and Bambu Slicer. Prusa: No bugs at all, full model has been sliced without any missing layers. BambuStudio: Missing Layers in undefined heights. unable to start a print like this.

SaltWei, 2022-12-29:
Please share the model for us to debug. Thanks!

SaltWei, 2023-03-02:
No update for a long time. Close at the moment.
Reopen and continue to check when model comes

closed 2023-03-02 by SaltWei

immenz, 2023-03-06:
Hi, maybe I have a similar/the same issue.
I want to print the following printer case: https://www.printables.com/de/model/356301-arc-for-bambu-lab-p1p/files

Slicing looks as following: 
[image]

I've observed this behaviour with other models, too. Resulting always in failed prints, when I try to print