bambulab/BambuStudio #955: External Service Progress Updates (Discord, Telegram, Slack)

clairmont32, 2022-12-26:
**Is your feature request related to a problem? Please describe.**
Octoeverywhere is one of many examples for services that offer interval or time based messages for print progress. This helps you collaborate about a print with friends, coworkers, or community members in addition to keeping tabs on your print job in case something fails within software or hardware.

**Describe the solution you'd like**
Offer a configurable notifications section to various external services. Configuration items could be webhook url, api key, or oauth sign in for the service and print status intervals. Key events such as start/stop/pause/finish or user intervention required could also be selectable items to send a notification to a service.

**Describe alternatives you've considered**
I'm new to Bambu products but they seem pretty closed sourced so I dont have a potential alternative.

Bambu Studio v 1.4.0.18 w/ X1CC

fang64, 2022-12-26:
@clairmont32 might not be what you are looking for but there is a MQTT interface, https://community.home-assistant.io/t/bambu-lab-x1-x1c-mqtt/489510

clairmont32, 2022-12-27:
Thanks I'll check this out more tomorrow. At first glance it seems like a simple pub-sub message handler?

fang64, 2022-12-27:
@clairmont32 yup it seems MQTT is pretty standard, but I suspect it's subject to change.

clairmont32, 2022-12-27:
Yeah officially supported solutions are always more preferable. Thanks for the lead.