bambulab/BambuStudio #957: Please bring back hybrid supports

davidjdixon, 2022-12-27:
**Is your feature request related to a problem? Please describe.**
The new tree supports are great, but in places where the support height is very small, they don't work well in my limited experience. Instead, the supports fail very quickly because there is not enough surface area being generated (resulting in consistently failed supports). Normal supports could solve this, but when supports take up a larger surface area, they use up a lot of room, and I find they collide with the main models, especially with a support brim.

**Describe the solution you'd like**
Hybrid supports were great for this application, where it would provide a partial tree from the build plate, away from the model but also give strong support under the model where the support is needed.

An example of a model that shows this problem can be found here: https://www.printables.com/model/125073-aa-and-aaa-battery-dispenser-for-skadis/files#preview

jukefr, 2022-12-27:
Was looking at commits because this is a banger feature and apparently its still there but under a new very obscure name.
https://github.com/bambulab/BambuStudio/commit/29dbc77e913ab48ed27d11504118388f0ed03fb3
`3. Remove hybrid(auto) type and use "Tree Hybrid" style to represent it.`

davidjdixon, 2022-12-30:
Oh, that's a weird place to put those options, especially as you can select "tree slim" and "tree hybrid" etc when using "default" support - this would surely be a bug to simplify/sanitise these options.

tangketan, 2023-03-05:
@davidjdixon Thanks for reporting that. Now the tree styles are only available under tree type support after version 1.5 (to be released soon).

closed 2023-03-05 by tangketan