bambulab/BambuStudio #958: Make color painting behave like Super Slicer's enhancement to Prusa - painted colors aren't just one surface layer

sprior, 2022-12-27:
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I tried Bambu Studio's color painting feature for a print and found that the painted color is only one layer/perimeter deep which when printed allows the underneath color to affect the saturation of the surface color.  I would suggest that this is rarely what the designer wants.
[image]

**Describe the solution you'd like**
I tried this in Prusa Slicer and found it behaves the same as Bambu Slicer - the painted color is only on the surface.  I then tried it in Super Slicer and found that the painted color is applied to the full depth of the feature (see screen shot).

I would suggest switching to Super Slicer's implementation of color painting.

**Describe alternatives you've considered**
There appears to be no way to color the model in Super Slicer and export the model 3mf file to Bambu Studio because it would only be in the project 3mf file.  The only other alternative would be if the user has the original model and can export it as a multi part body but this isn't available if you're starting from a model you didn't design.

**Additional context**
[image]

Add any other context or screenshots about the feature request here.
[image]

randyre, 2022-12-27:
I actually use the single layer a lot as it reduces filament switching, but not for every color. What I would like to see is both options available as I have also seen where some colors could use more layers or a particular model benefits from a different way of painting. 

maybe a box in the coloring tool to allow us to input how many layers the painting will affect ort something like that.

CameronBigger, 2023-02-02:
@randyre I think your idea of adding a "layer paint depth" field like you describe would make everyone happy. It could default to '1' and work like it currently does, but would allow pretty easy customization of how deep it should go for when surface painting isn't enough.

medic6578, 2023-03-26:
I have run into this myself. What would be nice would be allowing me to choose how many layers deep I want the color to go. So a couple of check boxes for 1 layer or all layers and a slider to choose the layer depth if I want to go that route.

gitjpk, 2023-03-27:
+1 on this issue. Please allow us to decide the number of layers

Ing-Dom, 2023-03-30:
this would be a great feature! +1 !

woodenforest, 2023-04-12:
+1 i really need this feature!

randyre, 2023-04-12:
I agree. Here is a picture with the standard painting. Really needed a couple more layers to make it opaque.
[image]

Britford, 2024-08-19:
+1 This feature would make the AMS much more useful.