bambulab/BambuStudio #959: Scaling percentage returns to 100%

randyre, 2022-12-27:
**Describe the bug**
After scaling an object by a certain percentage the scale in the menu returns to 100%

**To Reproduce**
Steps to reproduce the behavior:
1. open a file
2. go to scale
3. enter a scale percentage with snap scale on
4. object size changes as expected, then percentage value returns to 100%

**Expected behavior**
Scale percentage should remain at whatever percentage was input in relation to the original object size.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 10]
 - Version [1.4.0.18]

randyre, 2022-12-27:
As a side note, the reason this is important is twofold:

First is this is the way other slicers do it so reduces confusion,

Second reason is when you go to print more parts and import a file you can remember what scale you used. Currently I'm saving the projects with the scale included in the file name just to make sure parts fit.

randyre, 2022-12-29:
One more issue with scaling this way.  Say you want to scale a part down to 98% to adjust fitment, then after printing you find you need to adjust down another 1% to 97%, but with the scale reset to 100% and then you need to adjust down another 1% from what's displayed as 100 but is really 98%, you can enter 99%, but that's a 1% reduction of what is really 98%, you're not getting 1% reduction or a 3% overall from the original size but rather something like 2.65% reduction due to scaling errors.

the only way around this would be to re-load the original object and scale to 97% to get the correct 3% reduction, but this is the long hard way around.

randyre, 2022-12-30:
Was just using scaling for some dowel pins I was printing and noticed scaling worked as intended, I was stumped and when back and looked and realized i was scaling multiple items at once, so made multiple pins and found the same behavior. So seems to be a but when multiple items are selected and then scaled.  Will amend the original bug report.

randyre, 2023-01-03:
May be related to #988

randyre, 2023-01-10:
@zhimin-zeng-bambulab   When importing multiple files, they are at 100% of their imported size. If I scale one to 120% then it should display 120%. Then if I select all of them and scale them to 150% it should still show 150% as the intended outcome was to scale all items to 150% and the original scaling of 120% of a single object would no longer be relevant. This is also how other slicers do it, so why are we trying to do thing that make it hard to re-learn behaviors.

Or is there a way to perform some logic that if select all objects and one of them is no longer at 100% that the default behavior is to set scale the way it does currently, but if all objects start as 100% then scaling percentage stays at that scale?

zhimin-zeng-bambulab, 2023-01-11:
Different software has some behavioral differences, most of our behavior is consistent with Prusa, the scaling behavior is also consistent with Prusa.
When selecting multiple objects, it only display the first selectd obejct's value, this bevavior is also not very reasonable.

randyre, 2023-01-13:
@zhimin-zeng-bambulab 

 Thanks, I've never used Prusia Slicer or a derivative until now. All I know is that it makes scaling large multi-file projects much harder when it doesn't retain the scaling percentage from the original file. 

I guess a workaround is to import all files into Cura to scale them as needed, then save them to a separate folder before importing them into Studio for printing. It's a lot of extra steps but at least I don't spend 10-15 hours printing a part only to realize it's not been scaled properly for the project.

zhimin-zeng-bambulab, 2023-01-31:
We will optimize this behavior in the future, if we find a better way, but for now we will keep the current behavior. Thank you for your suggestion and for your understanding!

closed 2023-02-01 by zhimin-zeng-bambulab