bambulab/BambuStudio #961: Object/part setting filament change doesn't work any more

Gridzilla, 2022-12-27:
**Describe the bug**
Object/part setting filament change doesn't work any more

**3mf File for This Bug**
Any object

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Object/part setting'
2. Click on 'Filament dropdown'
3. Select eg. filament 3
4. The dropdown reverts to filament 1

**Expected behavior**
It is expected to be able to change the filament setting

**Screenshots**
100% reproducible, when using 1 AMS

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.4.0.17

Gridzilla, 2022-12-27:
[image]

c2h5oh, 2022-12-28:
Can't repro - I've been printing filament testers for all my filaments for the last day or so and this has worked 100% of the time

Gridzilla, 2022-12-28:
[2022-12-29 00-16-54.zip](https://github.com/bambulab/BambuStudio/files/10316864/2022-12-29.00-16-54.zip)
Short video demonstrating the issue

SaltWei, 2023-01-11:
> We have fixed this issue in the next version。

Maybe the next next version?

closed 2023-03-02 by SaltWei