bambulab/BambuStudio #963: [Feature Request] Add hardware acceleration

TheCableGuy99, 2022-12-27:
**Is your feature request related to a problem? Please describe.**
I have a very powerful laptop and when slicing some things it takes an abnormal amount of time. Once sliced moving the build plate around and zooming in and out etc... is all jerky. My guess (excuse me if I'm wrong) but it's using the CPU or the CPU's graphics card rather than the dedicated graphics card.

**Describe the solution you'd like**
In most programs I see they give you an option to select the graphics card to do the processing as it's far more powerful and capable for these kinds of tasks. I believe they call it "Hardware Acceleration". I think this would be a great alternative.

**Describe alternatives you've considered**
I haven't really. This PC is more than capable of running this program and shouldn't be having any issues with lag, jerk and slowdown when slicing or viewing the preview.

**Additional context**
Add any other context or screenshots about the feature request here.

randyre, 2022-12-27:
This is basically the same issue as #774 . It really needs to be addressed as complex models are near impossible to look at prior to printing.  I have a Nvidia RTX3090 and it behaves the same as my low-end workstation, so there is an issue with graphics acceleration.

randyre, 2022-12-27:
I would also like to see the label changed to "bug" as this almost makes the slicer unusable on large complex models.

TheCableGuy99, 2022-12-28:
> This is basically the same issue as #774 . It really needs to be addressed as complex models are near impossible to look at prior to printing. I have a Nvidia RTX3090 and it behaves the same as my low-end workstation, so there is an issue with graphics acceleration.

I agree in that it appears to be complex models more than larger models. I will leave it down to the moderators if they wish to keep it as a feature request of bug though.

lanewei120, 2022-12-29:
> **Is your feature request related to a problem? Please describe.** I have a very powerful laptop and when slicing some things it takes an abnormal amount of time. Once sliced moving the build plate [...]

are you using dual graphics card?
if yes, please help to show the two graphics card.

could you have a check, when doing the preview
which gpu is used?
[image]

TheCableGuy99, 2022-12-29:
Hi,

It appears it is using "GPU 0" which is the CPU's instead of "GPU 1", the dedicated nVidia GPU confirming our suspicions.

Please see the attached images:
Hardware Specs.png: Showing I have a more than powerful enough PC to run this smoothly.

Slicing Process.png: Showing it's using "GPU 0" while slicing

Preview.png: Showing it's still using "GPU 0" while previewing the item. I was actually holding the plate and moving it around as I took the screenshot to be sure "GPU 1" didn't kick in at any point and it didn't.

It's slow to slice complex models and preview them because it's not using the dedicated nVidia GPU 1.

Thanks.

[image]
[image]
[image]

lanewei120, 2022-12-29:
@TheCableGuy99 
could you select "nvidia graphics card" for bambu studio for a try?
[image]
[image]
[image]

after select "nvidia graphics card", you can use the task manager to confirm
and check the performance difference

TheCableGuy99, 2022-12-29:
I run Windows 11 and there’s a way to do it in the settings so did it that way before creating the thread. It didn’t seem to make much difference… if any at all.

lanewei120, 2022-12-29:
after setting, 
from task manager, does it using gpu 1?

TheCableGuy99, 2022-12-29:
> after setting, 
> 
> from task manager, does it using gpu 1?

Admittedly, I didn't check task manager after trying, I just noticed the performance wasn't any better.

I'm out right now so will check shortly and let you know to confirm it does actually use GPU 1 when told to.

TheCableGuy99, 2022-12-29:
@lanewei120 Okay I've had chance to check now.

Adding it in nVidia control panel doesn't seem to force it to using GPU 1. It still uses GPU 0 for slicing and previewing.

However, going to "Windows Settings > System > Display > Graphics" and adding Bambu Studio in there allows you to set it to use the nVidia Graphics Processor in the same way it should work in nVidia Control Panel. When doing it this way it seems to flick between processors.

It does appear to use it for slicing according to Task Manager but it doesn't seem any quicker. The preview also appears to be using GPU 1 now but doesn't appear to be much better.

I wonder if Bambu Studio needs to be optimised in some way to use the GPU 1 better for performance? It should also have the option which GPU to use within the program itself ideally.

TheCableGuy99, 2022-12-29:
Another thing to note that I've just found is that it crashes a lot when set to use GPU 1. And the Camera Live View will not load.

lanewei120, 2022-12-30:
it is slow especially under preview for big model
it seems using the dedicated gpu can not solve this issue

we will try to optimize it to use some other rendering method
however, I am afraid we don't have good ideas currently

TheCableGuy99, 2022-12-30:
Hi,

I suppose all we can ask is if you can do the following:

1) Add the option so the user can select between CPU or GPU processing (and remember the setting when the application is closed). This should only be needed for slicing and previewing.
2) Check the stability of the program and previewing of the camera works while this is enabled and resolve any issue why not (as it currently causes these problems when forced into GPU mode).
3) Any idea's you can to optimise the program to either be smoother natively and with the GPU being used.

Those are my requests and thank you for listening :)

lanewei120, 2022-12-30:
> Hi,
>
> I suppose all we can ask is if you can do the following:
>
> 1. Add the option so the user can select between CPU or GPU processing (and remember the setting when the application is closed). This [...]

to use CPU or GPU, it is decided by the API used in slicing and preview
for Preview, as we use OPENGL to render the lines which will use GPU definitely
for slicing, as there are lots of geometry computations, which will use CPU

we can not optimize the performance by choosing CPU or GPU

we will try to do more analysis  in preview and think some way to optimize the performance

TheCableGuy99, 2022-12-30:
"for Preview, as we use OPENGL to render the lines which will use GPU definitely"

I'm not convinced about this as I tested it before and it was using GPU 0 which is the CPU.

Anyway, I'll leave it with you, I'm sure you'll do all you can to help.

Thanks again.

TheCableGuy99, 2022-12-30:
@lanewei120 

Sorry but I just wanted to add that I have just tested the same model duplicated 10 times (as per my testing before) in different slicers.... Cura, SuperSlicer and Simplify3D and all of them completely out perform Bambu Studio. They slicing is fine (They also show a progress bar which is good) but the preview is so smooth. Simpligy3D is the worst, it's slightly laggy but only slightly and as said is far better than Bambu Studio.

There must be something you can do if all these slicers are performing just fine?

lanewei120, 2022-12-31:
@TheCableGuy99 

thanks for doing this

how about prusaslicer?
for preview, could you give a video for us to better understand the cases?

we will look into the details

TheCableGuy99, 2022-12-31:
@lanewei120 

Okay I've done Videos with all the major slicers and Bambu Studio is by far the worst performer in slicing time and previewing the plate taking a whopping 100 seconds to slice. The next slowest took less than half the time of that at just 40 seconds and was Simplyf3D. The fastest was PrusaSlicer with an impressive time of just 11 seconds to slice. 

Please find below my results and a link to download the video (please let me know when you have the download so I can remove it):
[https://www.privatespace.uk/SlicerVideos.rar](url)

-=PrusaSlicer=-
Pre-Slicing Preview: Fine
Slicing Time: 11 seconds
Preview: Very fast and smooth

-=Cura=-
Pre-Slicing Preview: Fine
Slicing Time: 23 seconds
Preview: Very fast and smooth

-=SuperSlicer=-
Pre-Slicing Preview: Fine
Slicing Time: 35 seconds
Preview: Very fast and smooth

-=Simplify3D=-
Pre-Slicing Preview: Fine
Slicing Time: 40 seconds
Preview: Ever-so-slightly laggy but more than acceptable and usable

-=Bambu Studio=-
Pre-Slicing Preview: Fine
Slicing Time: 100 seconds
Preview: Slow, Laggy, Jerky

Thanks.

Ali229, 2023-01-08:
Bambu slicer is pretty damn slow compared to prusa slicer, even though it's based on it, I have to pull up models in Prusa for adjustments before I pull them in bambu slicer, hope this fixes soon.

lanewei120, 2023-01-09:
> @lanewei120
>
> Okay I've done Videos with all the major slicers and Bambu Studio is by far the worst performer in slicing time and previewing the plate taking a whopping 100 seconds to slice. The next [...]

@TheCableGuy99 
you can delete the video
Could you also share the 3mf for us for a debugging?

for the slicing slow issue, it is may caused by the option "Reduce infill retraction"
[image]
could you disable it for a try also?

for the detailed reason, we can give some explaination later

TheCableGuy99, 2023-01-09:
Hi,

Thanks for confirming I can delete the file.

I have now tested with "Reduce infill retraction" disabled and it appears to have taken longer to slice, not shorter. The total time with Bambu Slicer was 133 seconds now.

Please find attached the 3mf file as requested, thank you. (Please note, github wouldn't let me upload a .zip, .rar or .3mf so I renamed the file to a .txt to allow me to upload it. Once you have downloaded it just rename it back to a ,3mf).

[Slice Test.txt](https://github.com/bambulab/BambuStudio/files/10373895/Slice.Test.txt)

Thanks.

lanewei120, 2023-01-10:
@TheCableGuy99 

we have root caused these two issues:
the slicing slow is caused by some tricky logic when loading gcode into preview, 
the rendering slow is caused by the imgui rendering blocked by some wrong finding logic
we will solve these two issues in next release

thanks for your help

TheCableGuy99, 2023-01-10:
Excellent, i'll look forward to that release. Thanks for all your help :)

TheCableGuy99, 2023-01-20:
Hi,

I noticed this was referenced in the latest release so tested it again and wanted to give you some feedback.

Firstly a huge increase in slicing time, well done!!! I was able to slice the same file in 40 seconds as opposed to the previous 100 seconds. But you might want to still look into that? It's still only on par with the slowest comparison, Simplify3D. Considering this is based on Prusa Slicer which does it in an impressive 11 seconds there's something not right in my opinion. 

However, the preview is still laggy/jerky.

Thanks.

lanewei120, 2023-01-20:
yes, we will continue to check the slicing performance

for the preview performance, have you did some comparisons?
we tested with your 3mf, the previous version is about 5 fps
and now is about 30 fps

lanewei120, 2023-01-20:
> would you mind if I sent you a privat

sure, Glad to receive your private message
thanks for doing this~

TheCableGuy99, 2023-01-20:
Thank you, I wasn't aware though that you cannot send private messages on Github (just tried and couldn't find a way so Googled it).

Is there any way you can share your email address (or other contact method) and remove it immediately? I'll get the email with it in and then you can remove it so it's not public.

lanewei120, 2023-01-20:
My email address: lane.wei@bambulab.com

TheCableGuy99, 2023-01-20:
Got thank you, i'll email you now if you want to remove it :) Thanks so much.

julie777, 2023-02-22:
I just want to add some info related to this. I also use a laptop with a discrete graphics card. When I started using Fusion 360 almost everything became unusable. The interface for everything was unbelievably laggy.

I gained a huge improvement by disabling Nvidia Container processes from running. I still have a few stutters here and there but is orders of magnitude better.

While diagnosing my problem I changed settings to force Fusion 360 to use performance setting for the GPU which caused it to use the Nvidia discrete graphics.

While trying everything that I could I found out that Microsoft has removed the hardware acceleration option from Office 365 applications. I found documentation discussion how they were moving away from applications specifically using hardware acceleration and that the Windows graphics system is supposed to be optimizing graphics implicitly now.

M3HNGRY, 2024-03-25:
Performance still slow on large or complex files. Please add GPU acceleration to the slicer...