bambulab/BambuStudio #966: Send To Printer

TheHogarthWorkshop, 2022-12-28:
when using underscore in name of file it reverts to saving as untitled name plate.

Example.
test_print

TheCableGuy99, 2022-12-28:
I was just about to report that it doesn't name them at all. However, I am using spaces and haven't tried a single word. Clearly an issue as you say though with naming plates and sending to the printer. See attached images...

[image]
[image]

TheHogarthWorkshop, 2022-12-28:
Yeah, I just had a weird one where the project was called Print_Object and I sent it to print, and changed its name to a single word, and it sent the original name with the _ so yeah, it's more than just that - theres something just not right there

TheCableGuy99, 2022-12-28:
Just to confirm I tried another without spaces or underscores and called it "Lids", it stills set it as "Untitled_plate_1", so not a spacing or underscore issue, it's clearly with naming as a whole.

TheHogarthWorkshop, 2022-12-28:
> Just to confirm I tried another without spaces or underscores and called it "Lids", it stills set it as "Untitled_plate_1", so not a spacing or underscore issue, it's clearly with naming as a whole.

yes. same here macOS M1

TheHogarthWorkshop, 2022-12-28:
Okay so I think I narrowed this down to the following..... 

It doesn't update the name field until you click OFF of the edit box. So following editing the name, you need to then click on a blank pice of space so it looses its green border, THEN click on SEND.

Guessing the 'save' button just needs some additional code there to ensure it captures the changed field without user having to make sure to click off it.

TheHogarthWorkshop, 2022-12-28:
Better now I know, but took me a while to realise, and also isn't intuitive anyway, needs to be like my comment above.

TheCableGuy99, 2022-12-29:
Hey, I just tested it as I normally press "Enter" and it's looking like mine works as well with this method (I tried with spaces in the name as well). I just clicked in the popup box elsewhere instead of pressing "Enter" and it worked.

That's a good bit of investigative work.... nice one!

closed 2023-08-02 by walterwongbbl